iNavFlight/inav #9850: Can't change altitude in Mission control

Vins1417, 2024-03-24:
## Current Behavior
Altitude cannot be changed if Sea level Ref is checked. When you enter a value higher than that indicated by default, the following message appears.

[image]

## Steps to Reproduce
Go to Mission Control tab
Add a point
In the point settings, check Sea Level Ref and then increase the Alt value

## Expected behavior
A calculation of the correct elevation...

- FC Board name and vendor: SpeedyBee F405 Mini
- INAV version string: 
# version
# INAV/SPEEDYBEEF405MINI 7.1.0 Feb 14 2024 / 15:55:53 (49325426)
# GCC-10.3.1 20210824 (release)

breadoven, 2024-03-24:
This is a Configurator issue already covered by https://github.com/iNavFlight/inav-configurator/issues/1987. Should be fixed now.

closed 2024-03-24 by breadoven

Vins1417, 2024-03-24:
I reinstalled the 7.1.0 RC1 configurator and it doesn't work..

breadoven, 2024-03-24:
The fix isn't included in RC1. It'll be included in the next release of Configurator.