iNavFlight/inav #9852: For MavLink 2 send RC_CHANNELS packet with 18 channels instead of RC_CHANNELS_RAW with only 8 channels

mvmn, 2024-03-24:
## Current Behavior
MavLink RC and MavLink protocol v2 has been implemented in INAV, but INAV still sends back RC_CHANNELS_RAW message which supports 8 channels only - even though MavLink RC supports 16 channels with MavLink v2, and 8 channels with Mavlink v1.

## Desired Behavior
Send RC_CHANNELS with 18 channels instead of RC_CHANNELS_RAW with only 8 channels - in case Mavlink v2 protocol is selected

## Suggested Solution
There is an existing PR already: https://github.com/iNavFlight/inav/pull/8282

## Who does this impact? Who is this for?
This is for anyone getting MavLink telemetry from the drone

MrD-RC, 2024-03-24:
Seeing as a PR already exists for this. It’s a bit pointless to raise a new issue

closed 2024-03-24 by MrD-RC

mvmn, 2024-03-24:
The PR isn't merged, isn't assigned to anyone - seems it's completely ignored without having a corresponding ticket/issue.

Since the PR isn't merged - this issue absolutely cannot be considered completed.

MrD-RC, 2024-03-24:
This issue can be considered pointless. There is no need to raise an issue for something that already has an in progress PR. It doesn’t need to be assigned to anyone. I’ve added a milestone so that the merge should be picked up.