iNavFlight/inav #9853: WING FC boards won't ARM

mickey2011, 2024-03-24:
Board: WING FC/FF35 version 0
Configurator: INAV 7.0.1
INAV 7.0.0 and the later versions 6

Trying to set up as a simple 'Wing FC' for a small 'Z84 Wing' i.e. 1 motor, 2 servo's, GPS, No FPV.

Issue is that they will not ARM, configurator 'setup' reports the NAVIGATION is not safe.

CLI ARMING Status reports "CLI NAV"

Many, many experiments show that if ANY 'NAV type MODEs are 'configured' (only configured, not activated) in the MODE setup/CLI then they won't ARM

Removing ALL NAVE MODES , sometimes allows ARMing but normally startover with a new/fresh INAV install and it will ARM, so long as NO 'NAV modes are configured.

Features.... GPS, No OSD, FrSky SBUS RX, Telemetry using SOFTSERIAL, Magnometer deactivated.

I have 3 of these boards with the same issue, the identical setup in MAtek WMN and SpeedyBee Wing boards works fine.

b14ckyy, 2024-03-25:
If nav modes are configured or rth, you need a valid GPS fix to arm. This is normal because otherwise you have no home position.

mickey2011, 2024-03-25:
Yes I am aware of this, this issue exists with a valid GPS fix, I can
normally get 8 plus Sats

b14ckyy, 2024-03-25:
Can you attach a txt with your diff please. 
Do the other working setups use the same GPS module? What GPS module exactly is in each setup? 

Beside the configurable minimum satellite number limit, there is also a hardcoded maximum HDOP limit I think. Interferences, electrical noise or reflections (from carbon parts) can cause a high HDOP despite having enough satellites. I think, but I am not sure about that, the max HDOP is 5.5. You can check that in the GPS tab.

mickey2011, 2024-03-25:
Yes I will tomorrow. My GPS’s are all Beitain BN220 or 880, I have tried
both types and as I mentioned, my setups work just fine on both Matek and
Speedybee boards. I have three WING FC boards and would like to use them on
Simple aircraft like Wings.

Thanks for your help

breadoven, 2024-03-25:
Can you arm using arming bypass, i.e. Yaw stick right, when the Nav modes are configured ? (Setting `nav_extra_arming_safety`)

mickey2011, 2024-03-26:
ah, I don't think this is a feature in INAV 7.0.0 ???

mickey2011, 2024-03-26:
Hello
WINGFC INAV 7.0.0 'Diff All" and ARM status.txt files attached

Thanks
Cheers

# status
INAV/WINGFC 7.0.0 Dec  5 2023 / 09:43:11 (895a4f31)
GCC-10.3.1 20210824 (release)
System Uptime: 97 seconds
Current Time: 2041-06-28T13:04:00.000+12:00
Voltage: 11.07V (3S battery - OK)
CPU Clock=168MHz, GYRO=MPU9250, ACC=MPU9250, BARO=BMP280
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
Stack size: 6144, Stack address: 0x10010000, Heap available: 1728
I2C Errors: 0, config size: 9220, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 3, used = ADC 3
      RSSI : configured = none, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 1, used = none
System load: 5, cycle time: 1017, PID rate: 983, RX rate: 113, System rate: 9
Arming disabled flags: NAV CLI
OSD: not enabled
VTX: not detected
GPS: HW Version: UBLOX8 Proto: 18.00 Baud: 19200
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    Galileo 0/0
    BeiDou 0/0
    Glonass 0/1
    Max concurrent: 3
diff all

# version
# INAV/WINGFC 7.0.0 Dec  5 2023 / 09:43:11 (895a4f31)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# Timer overrides

# Outputs [servo]
servo 1 1000 2000 1500 -100

# safehome

# features
feature -OSD
feature MOTOR_STOP
feature SOFTSERIAL
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper

# blackbox
blackbox NAV_ACC
[... 229 lines cut ...]
set battery_capacity = 1300
set throttle_idle =  5.000
set nav_fw_min_thr = 1150
set nav_fw_launch_thr = 1300
set nav_fw_launch_idle_thr = 1150

# battery_profile
battery_profile 2

set throttle_idle =  5.000

# battery_profile
battery_profile 3

set throttle_idle =  5.000

# restore original profile selection
mixer_profile 1
profile 1
battery_profile 1

# save configuration
save

#

sensei-hacker, 2024-03-26:
You may want to try turning off the virtual airspeed sensor.

 It's saying navigation is unsafe, which indicates a sensor problem or a waypoint mission loaded, but I don't see a waypoint mission or problems reported with the physical sensors.

mickey2011, 2024-03-26:
Thanks for that, I went looking to turn it off, but couldn’t find how to do
it.

Can you advise.

Thanks

sensei-hacker, 2024-03-26:
On the Configuration tab.

[image]

In 7.0, the virtual pitot was turned on by default. That will be turned off again in 7.1 because of issues as you experienced.

b14ckyy, 2024-03-30:
If you still have issues, just reply here.

closed 2024-03-30 by b14ckyy