iNavFlight/inav #9867: Optical Flow using Matek 3901-L0X sensor unusable in RC1 on SpeedyBee35

terryfritz99, 2024-03-28:
****

## Current Behavior
Switching to Position Hold after activating Surface Mode causes instability in hover level and wild increases in altitude. I now have a battery size hole in my ceiling due to this.

## Steps to Reproduce

1.Arm and hover
2.Turn on Surface Mode
3.Turn on Position Hold
4.Happens indoor or outdoor on any surface

## Expected behavior
On activating Position Hold with Surface Mode the quad should sink to the height above the ground surface set by the throttle level.

## Suggested solution(s)
 Out of my pay grade.

## Additional context
  Was working in INAV 7.0.0 and problem disappeared when I went form RC1 back to the 7.0.0 version, Sorry about the version dump below; I had already restored 7.0.0 and had to edit the string to make it look like the RC
https://pastebin.com/cuX3MRKG

---

# version

- FC Board name and vendor:# INAV/ 7.1.0 Feb  18 2024 / 11:04:12 (895a4f31)
- INAV version string:SPEEDYBEEF405MINI

b14ckyy, 2024-03-28:
Can you please doublecheck with this build if it still happens? (Please only outdoors in this case) https://1drv.ms/f/s!Ar6QsGgoXRgqyPk8ttBS_Kp6Auy8xA?e=kf9aGf 

This is the final 7.1 version and also includes the configurator for Windows.

terryfritz99, 2024-03-29:
Installed the Version 7.1 release and test flew in Position and Altitude Hold modes which worked well considering our chronic wind  here. Next tried Surface Mode with Position Hold and it worked as well as the 7.0 version with terrain following from 100 up to 500mm. That is about as good as I can ever get it over short grass. I noticed other users mentioned Range Finders too so it looks like this version is OK to go for Optical Flow. Thanks for your help.

New Paste Bin address: https://pastebin.com/XkM6EdYK 
New Version: version
# INAV/SPEEDYBEEF405MINI 7.1.0 Mar 27 2024 / 16:08:51 (59a6ee61)
# GCC-10.3.1 20210824 (release)
New FC Board Details: version
# INAV/SPEEDYBEEF405MINI 7.1.0 Mar 27 2024 / 16:08:51 (59a6ee61)
# GCC-10.3.1 20210824 (release)

b14ckyy, 2024-03-30:
Glad it worked. The new Sensor fusion and filters do their job :)

closed 2024-03-30 by b14ckyy