iNavFlight/inav #9871: no sdcard found Beerotor F4

guerradr, 2024-03-29:
## Current Behavior
SD card: Startup Failed

```
INAV/BEEROTORF4 7.0.0 Dec  5 2023 / 09:37:23 (895a4f31)
GCC-10.3.1 20210824 (release)
System Uptime: 120 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.00V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
SD card: Startup failed
Stack size: 6144, Stack address: 0x10010000, Heap available: 1828
I2C Errors: 0, config size: 9455, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = none
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 4, cycle time: 1008, PID rate: 992, RX rate: 9, System rate: 9
Arming disabled flags: CAL ACC RX CLI
OSD: MAX7456 [30 x 16]
VTX: not detected
``` 

## Steps to Reproduce

1. Inserted freshly formated SD Card (32g and 8g)
2. Tried INAV 6.0.0 / 6.1.1 / 7.0.0 / 7.1 RC1
3. The SD Card works with Betaflight 4.3.3 (SD card: Manufacturer 0x74, 30587904kB, 07/2014, v2.0, 'USD  ')
4. Tried with sdcard_detect_inverted = OFF and ON

## Expected behavior
SD Card recognized

## Suggested solution(s)
look at BF differences on the Board? 

## Additional context
DUMP from INAV
```

# dump

# version
# INAV/BEEROTORF4 7.0.0 Dec  5 2023 / 09:37:23 (895a4f31)
# GCC-10.3.1 20210824 (release)

# 
``` 

DUMP from Betaflight
```
# 

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.26V, Core temp=54degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 3638, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked
GYRO=ICM20689, ACC=ICM20689
OSD: MAX7456 (30 x 13)
BUILD KEY: e91eb9aebdb5013b8df5737d1392cebf (4.4.3)
System Uptime: 40 seconds, Current Time: 2024-03-29T16:17:38.273+00:00
[... 1024 lines cut ...]
set yaw_rc_rate = 7
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end

# 
``` 

- FC Board name and vendor:
- INAV version string:
# INAV/BEEROTORF4 7.0.0 Dec  5 2023 / 09:37:23 (895a4f31)
# GCC-10.3.1 20210824 (release)

b14ckyy, 2024-03-29:
Can you please attach diff's as a file and not plain text? 

What model SD card is it? Can you format it again please and on Sector size (in Windows) select STANDARD (or maybe DEFAULT) instead of 32kB. then try again. 
32GB might not work at all but 8GB should.

guerradr, 2024-03-29:
I tried following SD Cards (all freshly formatted with official SD Card Formatter).

Transcend 32GB HCI
Transcend 16GB HCI UHC1
Noname 8GB
NOKIA 1GB

all of them work fine on Betaflight 4.3.3

guerradr, 2024-04-01:
The board crashes if I select Blackbox in INAV 7.1.0 Configurator and has to be rebootet until MSP connection can be reopened. I guess it's because the SD Card is not responding.

guerradr, 2024-04-01:
Solution found!
In CLI:
```

set blackbox_device = SDCARD

```

and it got recogniced:
```
SD card: Manufacturer 0x74, 30587904kB, 07/2014, v2.0, 'USD  '
Filesystem: Ready

```

The INAV Configurator 7.1.0 still shows "No dataflash chip found" but Blackbox Tab works.

b14ckyy, 2024-04-01:
The Dataflash not found is normal. Its only active when you have on board blackbox flash. Not for SD cards. 
But thanks for sharing that fix. Maybe someone can try that and maybe it helps the other 2 similar reports. Could be a bad parameter default.

b14ckyy, 2024-04-03:
btw one last question. Did you manually select the SD Card here before? 
[image]

I just found with another FC that if I set it to Serial Device there (as it was in your dump) the FC locks up as well and now I can't access the Blackbox side as well. This is indeed a bug. Not sure if configurator or firmware side.

closed 2026-09-25 by b14ckyy