iNavFlight/inav #9876: INAV Black box not working - Matek H743 slim V3 #9540

Hex-Davis, 2024-03-30:
This is a Reopening of this ticket as the issues was never solved. After this was marked as closed i decided to just wait to see if an up-date would come out to fix the issue. No such luck. This issue continues to persist. Attached is another file recorded 50% loop time (1/2),cycle time was 2500. this recording should be roughly 80sec long but is <2sec long. it appears to be the first 2sec of the flight.
something maybe related is when looking at the available data in the black box analyzer the X-axis values for frequency are "infinity Hz"??
[LOG00001.TXT](https://github.com/iNavFlight/inav/files/14809290/LOG00001.TXT)

Hex-Davis, 2024-03-30:
Here are 4 more flights with greater than 2min of flight time each.

[LOG00002.TXT](https://github.com/iNavFlight/inav/files/14813523/LOG00002.TXT)
[LOG00001.TXT](https://github.com/iNavFlight/inav/files/14813525/LOG00001.TXT)
[LOG00003.TXT](https://github.com/iNavFlight/inav/files/14813522/LOG00003.TXT)
[LOG00004.TXT](https://github.com/iNavFlight/inav/files/14813521/LOG00004.TXT)

b14ckyy, 2024-04-01:
Hi Hex. 
this is a known issue with logging right now. Especially the H7 is a bit troublesome. If I remember correctly, with F765 for example the fastest stable logging rate I was able to achieve was 500Hz. With H7 and F4 as well as F722 everything above 250Hz can cause issues. 

you can reduce the logging fields by disabling some you don't need to allow higher rates but for now it is how it is until a better SD driver implementation can be found. 

If you wanna do noise analysis, you can disable most fields except gyro and accellerometer data and it should be easy to run 1000Hz (actually overkill already). For pure flight recording I have my logs running at 30Hz on all my fixed wing craft.

Hex-Davis, 2024-04-03:
I will give this a shot! Thank you very much i have been beating my head against this so long now.

closed 2024-04-03 by b14ckyy