iNavFlight/inav #9877: IFLIGHT_BLITZ_F722_X1 (For Defender25): Blackbox not detected

axylink, 2024-03-30:
Hi,

I compiled firmware for an iflight Defender 25, and the FC internal Blackbox is not detected
I can also see lots of I2C Errors: 1081
Any idea?

# status
INAV/IFLIGHT_BLITZ_F722_X1 7.1.0 Mar 27 2024 / 12:48:49 (59a6ee61)
GCC-10.3.1 20210824 (release)
System Uptime: 911 seconds
Current Time: 2041-06-28T02:04:00.000+01:00
Voltage: 0.01V (0S battery - NOT PRESENT)
CPU Clock=216MHz, GYRO=ICM42605, ACC=ICM42605, BARO=SPL06
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=UNAVAILABLE
Stack size: 6144, Stack address: 0x20010000, Heap available: 1716
I2C Errors: 1081, config size: 10455, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = none, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 18, cycle time: 516, PID rate: 1937, RX rate: 9, System rate: 9
Arming disabled flags: NAV HWFAIL RX CLI NOPREARM DSHOTBEEPER
OSD: MSP DisplayPort: BetaFlight Compatability mode (HD) [53 x 20]
VTX: not detected
GPS: HW Version: Unknown Proto: 0.00 Baud: 57600
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    Max concurrent: 0

[image]

b14ckyy, 2024-04-01:
do you have the variant with an on board ANALOG OSD or the one without? 
they might have different SPI definitions as OSD and SD Card both run on SPI.

axylink, 2024-04-01:
Good question - was not aware there were 2 versions. I run the DJI O3 VTX. What's the visible HW difference between the 2?

axylink, 2024-04-01:
If that can help, I tried to flash the IFLIGHT_BLITZ_F7_AIO but the firmware could not be loaded (size exceeded)

axylink, 2024-04-02:
Flashing IFLIGHT_BLITZ_F722_X1_OSD will lead to frozen FC (no blinking LED and INAV cannot connect). So I assume I own the variant without ANALOG OSD, right?

No better if I write "set blackbox_device = SDCARD"

I have the option to CLI "set blackbox_device = SPSFLASH", but still filesystem not detected

[image]
[image]

axylink, 2024-04-02:
Under BF, I have this:
[image]

[image]

axylink, 2024-04-02:
under INAV:
[image]

b14ckyy, 2024-04-02:
OK It has on board flash and not an SD card. My bad. 
I am curious why this causes I2C errors as the flash should also be SPI. I'm out of ideas then. Hope one of the Devs can help.

aetherhawkfpv, 2026-10-07:
I'm at a point where idk what to do , this ic2 error 1081 got me mad nervous lol.

sensei-hacker, 2026-10-07:
> I'm at a point where idk what to do , this ic2 error 1081 got me mad nervous lol.

Is the number of errors INCREASING?
If it's increasing there is an ongoing problem.
If it goes to 1081 and stops, something isn't ready at boot, then comes up and works okay.