iNavFlight/inav #9884: OSD on SpeedyBee 405 Wing and Mini not working

suchitmehrotra, 2024-04-02:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior
I was on Inav 7.1 and update my firmware to 7.0 

## Steps to Reproduce

1.Update Configurator to 7.1
2.Update Firmware to Auto 7.0
3.Turn On OSD

## Expected behavior
OSD must be visible  

## Suggested solution(s)
Downgraded to 6.1.1 and it works as expected 

## Additional context

---

- FC Board name and vendor:
- INAV version string:

suchitmehrotra, 2024-04-02:
I am using Analogue for VTX

b14ckyy, 2024-04-03:
Can't reproduce. F405-Wing, Configurator 7.1 Final and 7.0 Firmware final. No Problem: 

[image]

b14ckyy, 2024-04-03:
Please describe your issue more in detail and upload a TXT file with a DIFF.

suchitmehrotra, 2024-04-05:
@b14ckyy  me update it again ans share the information.

b14ckyy, 2024-04-09:
Any update on a detailed issue description? OSD works just fine here.

closed 2024-04-17 by DzikuVx