iNavFlight/inav #9888: Aileron Selection

Sotm12, 2024-04-02:
## Current Behavior
Inav Wing when you select Airplane With Tail assigns 2 aileron Servo assignment with no option to only have 1 aileron output. 
## Desired Behavior
Inav Wing when you select Airplane With Tail assigns 2 aileron Servo output  with no option to only have 1 aileron output. Need to be able to select only 1 Aileron output so that you add someother function to an additional output!! 
## Suggested Solution
Re-program the firmware so that either 1 or 2 aileron outputs may be selected. there by allowing one additional other output servo to be used. This has been wanted since early versions of Inav, why hasn't this changed? 

## Who does this impact? Who is this for?
Who is this for?  All Fixed Airplane users, not everyone flys with "Dual" ailerons. Most use 1 channel and a "Y" harness for their ailerons!!! This would allow an additional numbered servo for other assignment!!! 

## Additional context

Sotm12, 2024-04-02:
You cannot select only one aileron!!!!!!!!!!!!!!!!!!!!!!!!!!

Duh!!!

From: Sensei 
Sent: Tuesday, April 2, 2024 9:32 AM
To: iNavFlight/inav 
Cc: Sotm12 ; Author 
Subject: Re: [iNavFlight/inav] Aileron Selection (Issue #9888)

If you want to change what that output is used for, you can change it. (marked in red).
If you want to delete it, you can delete it (marked in green).

image.png (view on web)

The presets are a starting point so you don't have to set up each output one-by-one, for the common cases.

sensei-hacker, 2024-04-02:
Seriously, try actually clicking where I have marked.
Either the red or the blue.  Try it.  See what happens when you click the button marked in green.
It works way better than yelling "duh!!"

Also, please consider whether you want assistance, or you want any future messages from you to be immediately deleted.  You can choose which happens by how you choose to respond.

MrD-RC, 2024-04-02:
> You cannot select only one aileron!!!!!!!!!!!!!!!!!!!!!!!!!! Duh!!! 

There is no need to be rude to people who are trying to help you! If you do what @sensei-hacker said. You would have what you want.

closed 2024-04-02 by MrD-RC

Sotm12, 2024-04-02:
My Bad.
It did not accept a different receiver input when originally tried. I.e. I wanted to assign RC Channel 10 to it for a nose wheel steering mix. After I have re-booted everything, Inav, transmitter, model selection on transmitter, receiver, it will now accept the channel selection. Thank you for your help. I’ll close the matter.\0/

Duane

From: Sensei 
Sent: Tuesday, April 2, 2024 9:51 AM
To: iNavFlight/inav 
Cc: Sotm12 ; Author 
Subject: Re: [iNavFlight/inav] Aileron Selection (Issue #9888)

Seriously, try actually clicking where I have marked.
Either the red or the blue. Try it. See what happens when you click the button marked in green.
It works way better than yelling "duh!!"

Also, please consider whether you want assistance, or you want any future messages from you to be immediately deleted. You can choose which happens by how you choose to respond.

Sotm12, 2024-04-02:
NOT BEING RUDE _ _VERY FRUSTRATED!!!!!!!!!!!!!!

From: Darren Lines 
Sent: Tuesday, April 2, 2024 10:30 AM
To: iNavFlight/inav 
Cc: Sotm12 ; Author 
Subject: Re: [iNavFlight/inav] Aileron Selection (Issue #9888)

  You cannot select only one aileron!!!!!!!!!!!!!!!!!!!!!!!!!! Duh!!!

There is no need to be rude to people who are trying to help you! If you do what @sensei-hacker said. You would have what you want.

Sotm12, 2024-04-02:
What ever, guess you have never been frustrated Huh?

From: Darren Lines 
Sent: Tuesday, April 2, 2024 1:51 PM
To: iNavFlight/inav 
Cc: Sotm12 ; Author 
Subject: Re: [iNavFlight/inav] Aileron Selection (Issue #9888)

  NOT BEING RUDE _ _VERY FRUSTRATED!!!!!!!!!!!!!!

Yes, it was rude! Now shouting at me only compounds that.

You were given the answer before the rudeness. So there was no need for the frustration.

It is a simple mixer. If you don’t want two aileron outputs. Just remove one of them. Contrary to your comment. Most use two aileron ouputs with INAV. Not a Y lead. The default mixer for an airplane with a tail has been set to the one that most setups will use. However, if it is not right for your particular setup. A different mixer could be chosen (there are 5 or so other pre-defined mixes for an airplane with a tail). Or the given one easily modified.

DzikuVx, 2024-04-03:
Adults should control their emotions and act accordingly to the situation. Your reaction was an example of lack of emotion control. For the future: please learn how to control your emotions and act as a grown up adult human being. Thank you