iNavFlight/inav #9894: Barometer Altutude telemetry not sending to the radio

tifoha, 2024-04-04:
## Current Behavior
I can't get Baro Altitude telemetry to my radio TX16S (EdgeTX 2.9.4)
I have a drone Betafpv Pavo20 with FC: F4 2-3S 20A AIO FC V1 (STM32F405RGT6) 
this board has barometer BMP280
RX: Serial ELRS 2.4G Receiver
RX firmware version: BETAFPV AIO 2400 RX ELRS 
I got 19 sensors reading, even Vspeed, battery, but no baro altitude

## Steps to Reproduce
I enabled telemetry feature
I clearly see the readings from baro at sensors page (so baro definetly working)
I was trying to delete all sensors and discover them again multiple times
I was trying to read telemetry mirror from the aux1, but still got all other telemetry readings except of Altitude

## Expected behavior

## Suggested solution(s)
I should be able to discover baro altitude in TX16S and into AUX Telemetry mirror

## Additional context
Here is the [DUMP](https://pastebin.com/njiSrR3u) of my settings

---

# version
# INAV/BETAFPVF405 7.0.0 Mar 25 2024 / 11:09:05 (118cd485)
# GCC-10.3.1 20210824 (release)

b14ckyy, 2024-04-04:
may I ask first where you got this Firmware from? 7.0 compiled on March 25th does not make much sense. This does not sound like an official INAV Target.

tifoha, 2024-04-04:
this is a build from manufacturer of FC. But I think I can move the target to the newer version of Inav later

b14ckyy, 2024-04-04:
If there is an official target, I suggest to try 7.1 first to make sure your issue is not already fixed in between. 
The commit seems to be the same as the official INAV so that should be fine. Not sure why BFPV is building it themselves if there is a target but well :D

tifoha, 2024-04-04:
what is the target name?
I pulled the lates master and didn't find the target for betafpvf405, there is just (BETAFPVF411, BETAFPVF435, BETAFPVF722) which one should I use for F405?

tifoha, 2024-04-04:
> If there is an official target, I suggest to try 7.1 first to make sure your issue is not already fixed in between. The commit seems to be the same as the official INAV so that should be fine. Not sure why BFPV is building it themselves if there is a target but well :D

I think that they will add this target later, cause it is in development right now.

tifoha, 2024-04-04:
They defined USE_BARO in [target.h](https://github.com/BETAFPV/inav/blob/acbf44e412be30bab4de3b0a79d8ccc30f2ebb68/src/main/target/BETAFPVF405/target.h#L62), so I guess altitude telemetry should be added to scheduler in  [crsf.c#initCrsfTelemetry](https://github.com/BETAFPV/inav/blob/acbf44e412be30bab4de3b0a79d8ccc30f2ebb68/src/main/telemetry/crsf.c#L534) and be processed by this method [crsf.c#processCrsf](https://github.com/BETAFPV/inav/blob/acbf44e412be30bab4de3b0a79d8ccc30f2ebb68/src/main/telemetry/crsf.c#L496) during telemetry processing cycle, but it doesn't work for some reason. And I don't know how to debug this code using USB, so I have no clue what is actually going on, in runtime :(

b14ckyy, 2024-04-04:
There is an inofficial target here: https://github.com/sensei-hacker/inav_unofficial_targets/tree/unofficial_targets/src/main/target

you could use that target file and build from the 7.1 Release branch with that to see if that works.

tifoha, 2024-04-04:
Hmmm, Cannot find a valid ref in 
[release_7.1.0](https://github.com/iNavFlight/inav/tree/release_7.1.0)
where has it gone?

b14ckyy, 2024-04-04:
That repository is not up2date. But the target file is there. you can clone the official repository and use the target file from the inofficial one. Mabye @sensei-hacker can help.

tifoha, 2024-04-07:
Ok, I got it. @b14ckyy  thaks for advice! I modify the 7.1.0 release locally and add BARO + VARIO sensor into CRSF protocol
Is there a pinio_box feature in INAV7 like in betaflight, so I can set one of the pin as a switch for led strip?

b14ckyy, 2024-04-07:
PINIO is enabled on target when available on the FC on a dedicated pin. If you need it on another FC that has no pad for it you could repurpose a signal pad.

tifoha, 2024-04-07:
So I have to compile pins mapping with firmware? How to set pin on/off on
one of aux switches?

tifoha, 2024-04-07:
So I have to compile pins mapping with firmware? How to set pin on/off on one of aux switches? In Betaflight I did it with user1 mode, how it works in inav?

b14ckyy, 2024-04-07:
that's unfortunately above my paygrade in terms of coding. A developer need to answer that.

upyopyo, 2024-04-16:
I have the same issue for no Barometer(Alt) in telemetry, although getting other signals like RxBt, RSSI, etc.
INAV config 7.1.0
FC firmware 7.1.0 [speedybee F405 WING]
Baro should be from onBoard Baro, no GPS
I tried configuration tab > sensors&buses > Barometer set to AUTO and SPL06

Because I am a beginner for INAV, I may mistake INAV setting. Please let me know how to fix, thanks!

`status
INAV/SPEEDYBEEF405WING 7.1.0 Mar 27 2024 / 14:34:15 (59a6ee61)
GCC-10.3.1 20210824 (release)
System Uptime: 371 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.00V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=ICM42605, ACC=ICM42605, BARO=SPL06
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
SD card: Startup failed
Stack size: 6144, Stack address: 0x10010000, Heap available: 1728
I2C Errors: 0, config size: 10467, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = none, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 5, cycle time: 1005, PID rate: 995, RX rate: 49, System rate: 9
Arming disabled flags: CLI
OSD: not enabled
VTX: not detected

# get baro
baro_hardware = SPL06
Allowed values: NONE, AUTO, BMP085, MS5611, BMP280, MS5607, LPS25H, SPL06, BMP388, DPS310, B2SMPB, MSP, FAKE

baro_cal_tolerance = 150
Allowed range: 0 - 1000

inav_use_gps_no_baro = OFF
Allowed values: OFF, ON

inav_w_z_baro_p =  0.350
Allowed range: 0 - 10

inav_baro_epv =  100.000
Allowed range: 0 - 9999

osd_baro_temp_alarm_min = -200
Allowed range: -550 - 1250

osd_baro_temp_alarm_max = 600
Allowed range: -550 - 1250

# `

sensei-hacker, 2024-04-16:
> I have the same issue for no Barometer(Alt) in telemetry, although getting other signals like RxBt, RSSI, etc.

RxBat and Rssi are coming from your receiver. As opposed though from the flight controller.
Most likely, you don't have telemetry properly wired and configured from the FC to the RX.

Please make your own issue, with a screenshot of your Ports page attached, and a picture of your wiring.

gioclava, 2024-07-20:
Same issue:
FC firmware 7.1.2 [speedybee F405 WING]
ELRS telemetry
I tried configuration tab > sensors&buses > Barometer set to SPL06
In telemetry baro Alt is missing and VSpd stuck on 0 m/s
In sensors tab I can see barometer and altitude working correctly

zvikaf, 2024-09-11:
Same issue 7.1.2 withe speedybee F405 wing and ELRS :-(

villamany, 2025-07-20:
updated to inav 8.0.1 from 8.0.0 using diff_all, modified barometer type to SPL06, the select again bmp280 and variometer telemetry start to work in my taranis screen, disconnect FC from computer and battery. Recconect battery and stop working again