iNavFlight/inav #9895: Inav 7.1 keeps giving runtime calibration

EuphoricFPV, 2024-04-04:
Inav 7.1 keeps giving runtime calibration arming disabled because I have even refreshed to the old firmware and same thing is my cdi  dump corrupt? I have calibrated all sensors multiple times please help!!!!

sensei-hacker, 2024-04-04:
Run time calibration occurs automatically on boot. It calibrates the gyro, baro, and airspeed sensor at power-up. So, not related to the accelerometer and compass calibration.

It requires that the aircraft remain still for a couple seconds when you power up, so that the gyro can calibrate to what no movement looks like.

Similarly, for the airspeed and baro to calibrate, it should be not TOO windy.

A problem that can happen is if there is an airspeed sensor set up on the Configuration page, but there isn't actually any airspeed sensor attached.  Check the Configuration tab and see if there is an airspeed sensor set.  Including the fake or virtual sensor.  If you don't have an airspeed sensor, it can be set to "None".

b14ckyy, 2024-04-04:
Additionally to what Sensei already stated, VIRTUAL Airspeed is actually fine. But if you have no analog airspeed is installed, you MUST disable ADC for the airspeed. It can be set automatically on some FCs if an analog airspeed input is available. But with no sensor attached the pad is free floating, giving irregular inputs and preventing the runtime calibration from completion. 

So if its set to ADC, change it to virtual or none to fix the issue.

b14ckyy, 2024-04-07:
as no further response given, I will close this.

closed 2024-04-07 by b14ckyy