iNavFlight/inav #9899: [Target Bug] FlywooF745Nano - Wrong Motor Resource.

ramiss, 2024-04-06:
## Current Behavior

- Critical: The motor on pin C09 is not the correct pin for the FC to ESC stack connector - making this target useless.
- Nice to have: All of the motors have the wrong assignment, which also makes it annoying.

## Steps to Reproduce

1. Flash the FlywooF745nano target on an F745 nano FC
2. Test Motors
3. Motor on pin C09 does not spin because it is actually on pin E09 (see below)

cli motor resources from iNav:
B00: MOTOR3 OUT
B01: MOTOR1 OUT
**C09: MOTOR2 OUT <- Wrong pin**
E11: MOTOR4 OUT

## Expected behavior
All motors should spin

## Suggested solution(s)

1. Correct the motor resources to the following pins (as confirmed in the working BF target and tested by me):

resource MOTOR 1 B00
resource MOTOR 2 B01
**resource MOTOR 3 E09 <- the motor on C09 needs to be on E09**  (perhaps this was a typo?)
resource MOTOR 4 E11

## Additional context
For reference, here is the correct resource assignments from BF:
resource BEEPER 1 D15
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 E09
resource MOTOR 4 E11
resource MOTOR 5 C09
resource MOTOR 6 A03
resource MOTOR 7 B04
resource MOTOR 8 B05
resource LED_STRIP 1 D12
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 D08
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 E08
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 D06
resource SERIAL_RX 3 D09
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 E07
resource I2C_SCL 1 B06
resource I2C_SCL 2 B10
resource I2C_SDA 1 B07
resource I2C_SDA 2 B11
resource LED 1 A02
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 4 E02
resource SPI_SDI 1 A06
resource SPI_SDI 2 B14
resource SPI_SDI 4 E05
resource SPI_SDO 1 A07
resource SPI_SDO 2 B15
resource SPI_SDO 4 E06
resource ADC_BATT 1 C03
resource ADC_RSSI 1 C05
resource ADC_CURR 1 C02
resource PINIO 1 C00
resource FLASH_CS 1 A04
resource OSD_CS 1 B12
resource GYRO_EXTI 1 E01
resource GYRO_CS 1 E04
resource USB_DETECT 1 A08

---

- FC Board name and vendor:
- INAV version string:
# version
# INAV/FLYWOOF745 7.1.0 Mar 27 2024 / 13:04:03 (59a6ee61)
# GCC-10.3.1 20210824 (release)

cliffkhew, 2024-04-12:
Ya,I have the motor 3 not working,waiting for the fix.

wterreb, 2024-04-29:
Same for me.   On my FLYWOOF745NANO Betaflight has the motor mapping as :

Betaflight ,mapping    <-  this is working
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 E09
resource MOTOR 4 E11

But INAV 5.1 defaults as << This is not working 
B01: MOTOR1 OUT
C09: MOTOR2 OUT
B00: MOTOR3 OUT
E11: MOTOR4 OUT 

So it does not work at all in iNAV.  When want to I spin motor 1, then its  motor 2 that spins.    And when I try to spin motor 2, then no motor is spinning.   Etc etc.    

If this working in INAV, I would have been using INAV long ago.  But it feels like every time I try INAV I run into problems like this.

mmosca, 2024-04-29:
> @mmosca can you have a look, you changed something around in [a7e2911](https://github.com/iNavFlight/inav/commit/a7e29116881e37fc2ee3338bacb2dbddb9df5b4c) that includes these 2 outputs as well. Do they have different versions or was that a mistake?

I think the nano has been broken ever since it was added. I need someone to confirm #9990 works for both the nano and normal one.

They both now match BF config, but I don't own either.

closed 2024-04-29 by DzikuVx

ramiss, 2024-04-29:
I'm happy to test the nano, if that's helpful.

wernerterreb, 2024-04-30:
Actually I was playing with this myself last night and  managed to get the motor assignment correct.   When I go to Outputs and spin the motors now, it is the correct motors that spin for all four.   But my pin assignment looks very differnt to what you committed:

Here is what I used:
This will work on a FlyWoo Explorer LR that has a FLYWOOF745NANO board.

Add this to the file /src/main/target/FLYWOOF745/target.c:

timerHardware_t timerHardware[] = {

   DEF_TIM(TIM8, CH2N, PB0,  TIM_USE_OUTPUT_AUTO,    0, 0),  // M1 , TCH_TMR8_CH2N       
   DEF_TIM(TIM8, CH3N, PB1,  TIM_USE_OUTPUT_AUTO,    0, 0),  // M2 , TCH_TMR8_CH3N       
   DEF_TIM(TIM1, CH1, PE9,   TIM_USE_OUTPUT_AUTO,    0, 1),  // M3 , TCH_TMR1_CH1DMA2_ST1        
   DEF_TIM(TIM1, CH2, PE11,  TIM_USE_OUTPUT_AUTO,    0, 1),  // M4 , TCH_TMR1_CH2

ramiss, 2024-08-06:
@DzikuVx - This is not resolved.  I just installed a fresh 7.1.2 [FLYWOOF745NANO] target on my quad and Motor 2 still isn't responding.  Is there any way to get this re-opened and resolved please?  How can I help?

b14ckyy, 2024-08-06:
@ramiss Please check the PR for the fix. Its targeted for 8.0 and did not go into 7.1.2
[image]

You can try a Nightly build of 8.0.

ramiss, 2024-08-06:
@b14ckyy - Excellent, thanks.  I'll try it!

ramiss, 2024-08-06:
Tested and confirmed fixed in nightly 8.0.0.
Thank you all!

PR #9990 (2024-04-29): Match F745 and F745NANO motor pins to BF target
INAV was using same motor pins for both variants.

This PR matches motor pins to BF.
Fixes #9899 

I don't own this fc, so can't verify the fix.