iNavFlight/inav #9906: Arteficial horizon not showing up in osd

udum6, 2024-04-07:
Hi, I have a problem with my inav plane. The problem is that my artificial horizon isn't showing up eaven after flashing the Flight controller. The rest of the osd works except from the artificial horizon. The fc is matek f405 wing v2. Thanks for  help.

MrD-RC, 2024-04-07:
What video system?

If analogue, did you upload the font to the flight controller?

udum6, 2024-04-07:
Yes system is analougue but what do you mean upload the font to the flight controller. Like upload the alingment of the board to the fc

MrD-RC, 2024-04-07:
On the OSD page. At the bottom there is the font upload options. It needs to be installed on the OSD chip. You should do this whenever you update.

udum6, 2024-04-07:
Thank you so much it works. You got yourself a subscriber.

closed 2024-04-07 by udum6