iNavFlight/inav #9907: Slow OSD (WTFOS) refresh rate

slackista, 2024-04-07:
All INAV versions suffer from slow refresh rate when using dji WTFOS osd. Rapid high amp changes is good indicator of bad PID/filter tune. With INAV osd amp meter is useless because of slow update interval. On average it looks like 3Hz. 
Changing `osd_msp_displayport_fullframe_interval` makes no difference.  
With betaflight refresh rate is way higher with the same default settings of the WTFOS.

MrD-RC, 2024-04-08:
In INAV MSP DisplayPort. The drawScreen function is called at 250Hz. Inside the drawScreen function, the rated is reduced to 62.5Hz. This is plenty fast enough. 

[image]

`osd_msp_displayport_fullframe_interval` only determines how often the screen os cleared and all the data sent again. This doesn't affect how quickly the OSD is refreshed. That is still at 62.5Hz. 

https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#osd_msp_displayport_fullframe_interval

There is possibly something in the OSD creation slowing it down. But the refresh rate of MSP DisplayPort is not a problem.

The blackbox would give better results for tuning.

b14ckyy, 2024-04-08:
Can you please send a screenshot of your OSD config? 
Just had a quick chat with @MrD-RC about it as well as I can confirm that OSD elements that are not the AHI or scroll bars are updated at a slower pace. 

it is possible that the VTX actively prioritizes elements. So that the necessary realtime elements like the AHI is updated at a very high rate but other elements with text and number updates are just sent every X intervals to preserve bandwidth. 

As a result, the more elements you have on your screen that need refreshing, the slower these elements will update. So compare your BF and INAV OSD layouts and see if you have more stuff running on INAV.

b14ckyy, 2024-04-08:
can you post a screenshot of both OSD setups please? Do you have more items enabled in INAV than in BF?

MrD-RC, 2024-04-08:
Just because the amps are not updating as quickly. Doesn't mean that MSP DisplayPort is updating slowly. That updates at 62.5Hz. It could be that other areas get priority to update. If it has the same update time on analogue, it's not MSP.

b14ckyy, 2024-04-09:
Overall I think there is nothing much we can do. The OSD is already updates as fast as we safely can and INAV uses already all the available bandwidth to not cause data errors. For tuning related issues it is much more helpful to use Blackbox logging. 

If you really want in flight monitoring then the fast updates of BF are also not very helpful as you can easily miss short spikes. A much better option would be to use programming. Something like this: 

[image]

This logic will store any peak current (with a measurement update rate of 50Hz) in `GVAR 0` and shows any spike of the last 1000ms (line 3). This will work even better than any high update rate OSD element. Just add GVAR 0 as an OSD element next to your current sensor to have a comparison. Just keep in mind that GVAR has no decimals so 15A wills how as a value of 1500. 

Something you can't do with BF.

closed 2024-04-09 by b14ckyy