iNavFlight/inav #9908: High FF and low mechanic respond oscillation. Need a way to lower FF respond speed

majianjia, 2024-04-08:
## Current Behavior

My wing (540mm wingspan) roll rate was really fast at 350 degrees per second, even though the roll mixer setting was just at 25%. But the servo speed wasn't keeping up, which made the auto-tune set a really high Feed-Forward (FF) number.

The PID settings are super low at 2, 2, 1, while FF is way up at 80.
Larger PID cause more wobbling. 

Because of this, the plane is wobbling a lot in Angle mode, but it's ok in Acro mode but really smooth in Manual mode.

I think the problem is the roll rate is high, but the servos are slow to react. How do I tweak the settings so the FF takes the slow servo reaction into account?

From the log I can see the FF is most significant weight of contributing the servo motion, which should be damped. 
I think a damper add to FF could solve this issue. 
The mechanics option only have Dterm and Iterm limit. I not sure it is releated but looks like where the damper should be?

[image]

[image]

[image]

INAV/MATEKF405TE_SD 7.0.0 Dec  5 2023 / 10:31:33 (895a4f31)
GCC-10.3.1 20210824 (release)

## Desired Behavior
Controller can handle the nimble plan with a (relatively) slow respond servo.

## Suggested Solution
A damper parameter for FF. 
The FF looks like just another P term for me. 

## Who does this impact? Who is this for?
For small and nimble wings

b14ckyy, 2024-04-08:
Feed forward has nothing to do with servo speed. if its autotuned too high, you did the autotune procedure incorrectly. In autotune you are supposed to do full stick input and hold it there for a moment. On a plane like this, it would mean one or two rolls each direction to autotune. Do not "wiggle" the stick. During the roll or pitch input, the servos are in a fixed position so their speed does not matter at all. 

Can you provide a log from your tuning flight?

majianjia, 2024-04-08:
Thanks for your reply @b14ckyy 
Here is the log. For the roll tuning, I did full stick for 1 or 2 seconds, many times. The number growth and then stop at a "optimal value". 
[LOG00041.zip](https://github.com/iNavFlight/inav/files/14903976/LOG00041.zip)

b14ckyy, 2024-04-08:
OK that sounds correct. Will have a look in the evening after work. Can you please also add a complete DIFF of your setup as a txt file so I can also check if a setting could be wrong

majianjia, 2024-04-08:
Thanks again, 
I'm curious about how feedforward (FF) is calculated in angle mode since FF traditionally relates to rate inputs. 
However, in angle mode, the input is the angle, or the difference of expected angle and the attitude of the wing. 
Shouldn't there be a way to set the expected rate of angle correction? For this particular flight, I had set the banking angle to 70 degrees. Does this setting influence the FF calculation in angle mode.

I dont have the exact PIDFF now since I manually tuned after this flight because it is still wabbling. 

The clostes backup is this one. 

```
# Mixer: servo mixer
smix reset

smix 0 1 0 -25 0 -1
smix 1 1 1 -50 0 -1
smix 2 2 0 -25 0 -1
smix 3 2 1 50 0 -1

# profile
profile 1

set fw_p_pitch = 9
set fw_i_pitch = 12
set fw_d_pitch = 10
set fw_ff_pitch = 80
set fw_p_roll = 3
set fw_i_roll = 2
set fw_d_roll = 3
set fw_ff_roll = 30
set fw_p_yaw = 3
set fw_i_yaw = 0
set fw_ff_yaw = 20
set max_angle_inclination_rll = 700
set max_angle_inclination_pit = 500
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.300
set nav_fw_pos_z_p = 35
set nav_fw_pos_xy_p = 70
set d_boost_min =  1.000
set d_boost_max =  1.000
set tpa_rate = 30
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 31
set pitch_rate = 14
set yaw_rate = 3
```

b14ckyy, 2024-04-08:
as a simple explanation: 

in Acro mode your stick input controls the rate setpoint in degrees per second and feeds it into the PIDFF Controller. the PIDFF output then goes to the mixer and then to the servos. 

In ANGLE the stick commands a setpoint angle relative to the horizon plane. That Commanded angle goes into a Simple Proportional controller (Angle Strength) that proportionally compares the setpoint angle with the actual attitude. This Angle P-Loop then outputs a Setpoint rate like in acro mode done by the sticks directly. 

Its the same for navigation modes, the chain just gets longer. Heading and altitzude errors create angle setpoints that create rate setpoints and it all ends up in the PIDFF - Mixer - Servo chain in the end. 

based on your profile settings I can see that your max bank angle is really high at 70°. it might be that with this high max bank angle the default Angle strength of 20 could be too agressive. you could reduce angle strenght down to 15 or 10 and see of that works for you. 

Also check how well the rate setpoint is tracked by the gyro in your log. I will have a look later on that to find out if your FF is overtuned or not.

majianjia, 2024-04-08:
hi @b14ckyy 
Thanks for the detailed explains. Sounds very promising to me!

Do you know the exact parameter name for angle p strength? i can't find similar in the setting. Thanks!

b14ckyy, 2024-04-08:
@majianjia 
[image]

majianjia, 2024-04-09:
@b14ckyy Thanks a lot!
I think that is the root cause of my wabbling issue. Seems like my PID was tune at a wrong direction. 
I will try to reduce the number by half and recover my PID in the next flight, thanks!

closed 2024-04-09 by majianjia