iNavFlight/inav #9910: iNav F405-WING v2 with CRSF receiver not indicating inputs on Receiver page

BywongFlyer, 2024-04-08:
## Current Behavior
I am using a new ERLS Radiomaster ER4 receiver with port 2 Tx & port 3 Rx connected to Matek F504v2 Rx3 and Tx3 respectively (x'd over Tx to Rx). Does not show receiver channels moving. I have confirmed the presence of CRSF data with a DSO for both Tx from the Receiver and TX from the F405. To ensure I had the correct UART mapping I confirmed the F405 TX3 output by turning on and off telemetry and confirm on DSO. I have confirmed that I have correct wiring from receiver Tx to RX3 input by changing the serial to SBUS and found that the iNav did then indicate expected receiver channel inputs. So SBUS works with my wiring. Is looking like iNav may be the issue however am not confident as many other users use CRSF with this hardware and iNav ok.

## Steps to Reproduce

1.Connect receiver Tx(chan2) to F405 Rx3
2.Connect receiver Rx(chan3) to F405 Tx3 
3.Config iNav for serial on UART3 and Telemetry on
4.Transmitter outputs set and expect to see on iNav Receiver page indication

## Expected behavior

Expected 
## Suggested solution(s)

## Additional context
FYI
set serialrx_provider = CRSF
set serialrx_inverted = OFF
Dump file:
https://pastebin.com/AFKJu0p3
---

- FC Board name and vendor: Matek F405-WING v2
- INAV version string: # INAV/MATEKF405SE 7.0.0 Dec  5 2023 / 10:29:45 (895a4f31)
# GCC-10.3.1 20210824 (release)

Radio: Radiomaster Pocket (FW:edgetx-pocket 2023-07-27)
Receiver: ER4 (ver: 3.3.0)

closed 2024-04-08 by BywongFlyer

OptimusTi, 2024-04-09:
Where is the CRSF v10 spec?

BywongFlyer, 2024-04-10:
I referred to a comment on another post. Best reference is GitHub.com/crsf-wg/crsf/wiki
Much debate on baud rate used by various FC.

⁣Get BlueMail for Android ​