iNavFlight/inav #9912: Continous Auto Trim active during maneuvers

b14ckyy, 2024-04-09:
## Current Behavior
If Continuous Servo Trim is enabled on fixed wings, it is constantly trimming even when not flying level or when flying slow or fast maneuvers. This leads to a few seconds of strongly out of trim plane behavior after long rolls or turns as well as loops and can in worst case cause a crash at a low altitude loop. It also causes a lot of false trim during landing when the pilot does a flare maneuver at low airspeed. 

## Steps to Reproduce
1. Enable Continous Auto Trim in Features
2. Fly in acro mode and fly level until plane flies dead straight with no stick input
3. Do a long continous bank 180° or more turn or fly a spiral
4. quickly level off and release the sticks
Result: the plane will be heavily out of trim so that a low I-Term cannot compensate anymore and it takes a few seconds to recover

## Expected behavior
Continous Servo Trim should only trim the plane if the rate setpoint is very close to 0 in Autolevel modes or within stick deadband in acro mode. The trim should be locked while the pilot or autopilot is doing any maneuvers, turns or altitude changes. 

## Suggested solution(s)
I checked the code as far as I was able to understand it and found that the Continous Servo trim has multiple sanity checks before it should get active. 

```
planeIsFlyingStraight && 
noRotationCommanded && 
planeIsFlyingLevel &&
sticksAreCentered &&
!FLIGHT_MODE(MANUAL_MODE) && 
isGPSHeadingValid() // TODO: proper flying detection
```
I think one or more of these sanity checks are not properly working. 

## Additional context
The issue must be in INAV for a long time. I just noticed it intensively while testing other stuff and investigated a bit more. Here are 3 flight logs I did for a test of https://github.com/iNavFlight/inav/pull/9905 on a 8.0 build: [LOG_2.zip](https://github.com/iNavFlight/inav/files/14916408/LOG_2.zip)

LOG 33 is with Continous servo trim enabled. You clearly see after a long turn, that Integral is pushing hard to keep the pitch rate at 0 and settles slowly as trim is corrected. 
LOG 34 I disabled CST and did a classic autotrim with the trim mode, clearly does not show this same 
LOG 35 is another flight after saving the trim and just redoing autotune again to validate the correct rates and FF with a persistent trim. 

---

- FC Board name and vendor: Matek F405-WTE
- INAV version string: INAV/MATEKF405TE_SD 8.0.0 Apr  8 2024 / 18:01:53 (55a16f67)

Yury-MonZon, 2024-05-06:
Few lines above the mentioned if statement we have:
````

ProfDrYoMan, 2025-04-17:
I did ask the same in the mentioned other ticket:

Any news on this?

Would love to keep servo auto trim enabled all the to ensure proper trim even if CG is moved or motor angle was adjusted.

This is a neat feature, but the flaw does prevent me from enabling it.

sensei-hacker, 2025-12-28:
This might do the trick, if someone wants to test it
https://github.com/iNavFlight/inav/pull/11215

sensei-hacker, 2026-01-02:
> Having a user-set rc channel determine whether or not it's active makes most sense to me.

One might ask, how that much different from regular auto trim?

Carticiak, 2026-01-02:
> > Having a user-set rc channel determine whether or not it's active makes most sense to me.
> 
> One might ask, how that much different from regular auto trim?

Including flap/airbrake engagement in the list of sanity checks for the autotrim to engage is what I was getting at.

b14ckyy, 2026-01-02:
@MrD-RC 

> But it's worth noting that it is a huge pain to set up and tune. As you can't just use trims on the radio. 

Nope. Just one flight. That's why we have blackbox. Can even use OSD values. 
Fly level, make sure plane is trimmed, no elevator mix. Switch manual mode. Deploy flaps and counteract with your pitch stick. hold it so the plane flies level again and you will have the exact offset value in the log or your OSD. 

Just add the mixer then and change the percentage with deployed flaps on the ground until the servo output for elevator matches what seen in the test flight. Done.