iNavFlight/inav #9918: INAV Black Box logging issue - H7 flight controller

Hex-Davis, 2024-04-10:
Hi Hex. 
this is a known issue with logging right now. Especially the H7 is a bit troublesome. If I remember correctly, with F765 for example the fastest stable logging rate I was able to achieve was 500Hz. With H7 and F4 as well as F722 everything above 250Hz can cause issues. 

you can reduce the logging fields by disabling some you don't need to allow higher rates but for now it is how it is until a better SD driver implementation can be found. 

If you wanna do noise analysis, you can disable most fields except gyro and accellerometer data and it should be easy to run 1000Hz (actually overkill already). For pure flight recording I have my logs running at 30Hz on all my fixed wing craft.

_Originally posted by @b14ckyy in https://github.com/iNavFlight/inav/issues/9876#issuecomment-2029474973_

b14ckyy, 2024-04-10:
Can you please also add a full dump as a TXT to your post. 
If even the workarounds don't work for you there is something fishy going on.

Jetrell, 2024-04-10:
>I have found the blackbox begins recording but then once throttle comes off idle ether the recording stops or the data becomes too acratic for it to record (loosing frames or maybe not registering them)

I find its the `blackbox MOTORS` field that causes most of the issues for SDIO logging.

Hex-Davis, 2024-04-11:
I will try not recording the motors. 
This is the first Dump which would line up with log004 above i think
[dump1.docx](https://github.com/iNavFlight/inav/files/14939264/dump1.docx)

Hex-Davis, 2024-04-17:
I have made some progress with this i have everything the same as the last dump but i changed the blackbox_denominator to 50 this way the loop time was still high but the recording interval was still low (160hz) i recorded every blackbox parameter one at a time and then did a few flights as well. the flights did not get recorded full as they were around 8 min each but the recording stopped early. Attached are each of those recordings labeled with the parameter monitored.

Hex-Davis, 2024-04-21:
My findings thus far: the low loop times and high Blackbox_denominator seem to give the best results but the higher the denominator the lower the frequency rang analyzed. My next steps are to use an external logger.

Hex-Davis, 2024-04-28:
My findings thus far: Using a 'Openlog Artemis' from SparkFun I have collected a few recordings but i need some assistance configuring things, not really my field. Any help would be greatly appreciated 

BBR = Blackbox ratio 
baud = is the peripheral baud rate in the ports tab 

[# Diff all.txt](https://github.com/iNavFlight/inav/files/15143250/Diff.all.txt)
[# dump.txt](https://github.com/iNavFlight/inav/files/15143252/dump.txt)

[Data0000_Baud250000_BBR_1over8_Looptime125.TXT](https://github.com/iNavFlight/inav/files/15143256/Data0000_Baud250000_BBR_1over8_Looptime125.TXT)
[Data0001_Baud115200_BBR_1over8_Looptime125.TXT](https://github.com/iNavFlight/inav/files/15143257/Data0001_Baud115200_BBR_1over8_Looptime125.TXT)
[Data0002_Baud115200_BBR_1over16_Looptime125.TXT](https://github.com/iNavFlight/inav/files/15143258/Data0002_Baud115200_BBR_1over16_Looptime125.TXT)
[Data0003_Baud115200_BBR_1over32_Looptime125.TXT](https://github.com/iNavFlight/inav/files/15143259/Data0003_Baud115200_BBR_1over32_Looptime125.TXT)
[Data0004_Baud57600_BBR_1over8_Looptime125.TXT](https://github.com/iNavFlight/inav/files/15143261/Data0004_Baud57600_BBR_1over8_Looptime125.TXT)
[Data0005_Baud57600_BBR_1over16_Looptime125.TXT](https://github.com/iNavFlight/inav/files/15143262/Data0005_Baud57600_BBR_1over16_Looptime125.TXT)
[Data0006_Baud57600_BBR_1over32_Looptime125.TXT](https://github.com/iNavFlight/inav/files/15143263/Data0006_Baud57600_BBR_1over32_Looptime125.TXT)

mmosca, 2024-12-08:
If you guys want to help test, this nightly build should have a fix for it: https://github.com/iNavFlight/inav-nightly/releases/tag/v8.0.0-20241208.162

closed 2025-01-19 by b14ckyy