iNavFlight/inav #9931: Servos Freeze when both stick moving, even when Armed using INAV 7.1 on Speedybee F405 Wing with FLYSKY i6X via IBUS

drxOne, 2024-04-11:
Thanks in advance for any assistance...

I have been trying to figure out this issue for over a week with numerous configurations and re-writing clean firmware to the Speedbee F405 Wing board, but nothing is resolving the issue.

Whenever both sticks of my FLYSKY i6x radio are moving simultaneously, it seems that there is some stick command being executed that causes the servos to freeze, even when it is in ARM mode.  When it freezes, I can see that if I started in "PID Profile 1" it will end up going to "PID Profile 3" as shown below:

[image]

This issue occurs even thought I have the "OSD" and "Profile selection with TX stick command" options disabled under the "Configuration" tab.  Please see the image below:

[image]

Thanks again.

drxOne, 2024-04-11:
I was able to reproduce the same issue in INAV 7.0.  However, when I tested the Speedybee F405 wing board with ArduPilot 4.5.1 the issue is NOT happening.

b14ckyy, 2024-04-11:
if it happens when armed as well, can you please record a blackbox log while armed and you reproduce the issue? 
It should be impossible to enter the CMS menu when armed at all or to do any stick commands. the CMS menu is the only thing that can fully freeze the servos. This is normal when disarmed.

drxOne, 2024-04-11:
> if it happens when armed as well, can you please record a blackbox log while armed and you reproduce the issue? It should be impossible to enter the CMS menu when armed at all or to do any stick commands. the CMS menu is the only thing that can fully freeze the servos. This is normal when disarmed.

Ok... I will do it and get back to you.  Just have the ArduPilot 4.5.1 on it now to complete a test for a project I am working on.  Should be done on Sunday.

MisterNybble, 2024-04-15:
Yup, Just thought I'd try a new FC (Speedybee F405 Wing) I was working through some acrobatics on the bench doing quick snap rolls mixed with quick rudder/elevator movements and boom...the FC and servos just locked up.  fs-i6 fs-a8s sbus and just 4 servos.  Very repeatable (not alway in the same place).  I've even had it happen after one simulated snap roll.  Weird.

drxOne, 2024-04-15:
> Yup, Just thought I'd try a new FC (Speedybee F405 Wing) I was working through some acrobatics on the bench doing quick snap rolls mixed with quick rudder/elevator movements and boom...the FC and servos just locked up. fs-i6 fs-a8s sbus and just 4 servos. Very repeatable (not alway in the same place). I've even had it happen after one simulated snap roll. Weird.
> 
> I'm wondering if this is a Speedybee F405 Wing issue otherwise there would be lots of crashed airplanes?
> 
> Am I accidently issuing stick commands?

Thank you very much MisterNybble for confirming my results.  I even tried disabling the "OSD" and/or "Profile selection with TX stick command" options, but it is still happening.

b14ckyy, 2024-04-15:
Was your flight controller armed and are you able to provide a Blackbox log if it was?

jankiefer, 2024-10-18:
Same behaviour here on a Speedybee 405 Wing Mini using ELRS receiver from jumper T15, unfortunately the blackbox does not work as sd card is not recognized.  Servos are usable for a while and then lock up, sometimes they do recover a while later but usually you habe to restart the FC. The weird thing is that I have two other similar setups working where I haven't enccuntered the issue (yet?)

b14ckyy, 2024-10-18:
Video please. Its very likely a stick command applied as well. When Armed it should not happen.

b14ckyy, 2025-01-13:
@sardelich please read my last comment.

Daturley, 2025-01-18:
Encountering the same thing.   Horus trans with Radiomaster ELRS module, Ver 8 rc3, Speedybee 405 Wing, Radiomaster ELRS receiver.   Switched to R9 reciever same occurrance, servos freezing if both sticks are stirred.  Seems to be easier to trigger the freeze when stirring right stick counter clockwise, left stick clockwise.   Tried different packet rates, stick commands on and off.   Sometimes the stick stir freezes the servos in the first 3-4 rotations, sometimes after 20 or so.   Stirring the sticks more slowly will produce a momentary freeze that recovers in a few seconds sometimes, rapid stick stir is unrecoverable most of the time.

Hope this helps......

Daturley, 2025-01-18:
Tried FRSky reciever eliminating the ELRS setup variable... same result.

sensei-hacker, 2025-01-18:
[image]

When not armed, pushing all four axis to maximum positions will issue stick commands.

b14ckyy, 2025-01-18:
Since we still have no valid confirmation that anything locks up when armed and the stick commands are the reason why it looks like it locks up when disarmed (as the FC enters the OSD menu or loads/stores settings) I will close this.

closed 2025-01-18 by b14ckyy

Daturley, 2025-01-18:
On my setup this was always when armed with strong gps signal and in autolaunch mode or flight mode.

b14ckyy, 2025-01-19:
Can you provide a blackbox log? None of the devs can replicate it, we don't have it on video and we have no blackbox log. If it happens when armed we need a log.

Daturley, 2025-01-19:
will get home to the unit Feb 4th will get a blackbox log to forward to you then.  thank you very much for looking into this.....

Daturley, 2025-01-19:
And a video of the fc status and control input.....

Daturley, 2025-01-19:
I am away from home until the 4th of February.  When I get back I will get
a focused black box file to forward to you....  along with video.....

Thanks for all you do for the furthering of this technology and the hobby
for all of us!
Dave

Daturley, 2025-01-31:
Regarding the freezing of servos when stirring the sticks, I thought the
board was armed, but it was in fact NOT armed so disregard and my apologies
for taking up your time....
Dave