iNavFlight/inav #9935: INAV 7.1 the "Battery left" value which I send as telemetry data (FrSky) to my Tx is not working anymore

Thomasa66, 2024-04-11:
INAV 7.1
The telemetry value for "Battery left "which I receive (FrSky) in my Tx starts everytime with 1 and the rate increases by one every few seconds . In OSD the  "Battery left" value is correct. For testing purposes: I reinstalled INAV 6.x and there the telemetry value I receive for  "Battery left" is correct .. reinstalled 7.1 again  and  the value is broken. 

Other telemetry values (like 5-6) which I need seem to be O.K. though I did not check all telemetry datas.

b14ckyy, 2024-04-12:
https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#smartport_fuel_unit

`set smartport_fuel_unit = PERCENT`

Phil-MC, 2024-04-14:
@b14ckyy  Thanks for the solution. I had the same problem with my TX16S II Edge Tx/ ELRS outfit.