iNavFlight/inav #9937: Option to send airspeed instead of gps speed for CRSF radios

juanxli, 2024-04-12:
## Current Behavior
There isn't an option for this and you are stuck with GPS speed on CRSF/ELRS on the radio screen 

## Desired Behavior

**Being able to change the source of the speed shown on CRSF radios ** between airspeed and GPS speed.

## Suggested Solution

Add a function that substitutes the airspeed instead of the GPS speed telemetry of the radio.
An option similar to  "osd_speed_source" but instead "telemetry_speed_source"

## Who does this impact? Who is this for?

Wings with CRSF/ELRS and a pitot tube/airspeed sensor without an FPV system 

## Additional context

This is because FPV isn't allowed in some clubs or legislations only Line of sight

SpidaTuSefen, 2024-10-29:
Frsky have this and lots of LOS pilots use it so their airspeed is called out on final approach when landing bigger, scale fixed wing.

SpidaTuSefen, 2024-10-29:
It would also mimic airspeed calls by co-pilot used in manned aircraft - very professional way to handle take off, climb and landing.

SpidaTuSefen, 2024-10-29:
FPV is only legal at MAAA clubs in AU which a few and far apart, not to individual pilots at public spaces. So LOS iNav is the only legal option. C Asa even consider a monitor to be FPV.

ShikOfTheRa, 2024-11-24:
This would be really usefull. I am surprised no more upvotes.
Flying with a glider and need to keep airspeed above stall speed. Cant use alarms on tx with gps speed because sometimes airspeed = 3-kmh and gps speed = 0kmh...

Possible to add it as cli feature? replace gps speed with air speed when sending over crsf?

dcan999, 2025-05-12:
Did this go anywhere? Didn't see anything in 8.x release but would really like it. When I lose video I like to monitor the RTH using the radio telemetry and would like to see the airspeed rather than ground speed to know if I need to override the throttle etc.

viennalex76-alt, 2025-12-17:
@JetsRfaster You're right and your issue is absolutely relatable because iNav itself doesn't send any Airspeed data via CRSF, that's why you don't see any airspeed data on EdgeTX. The ELRS protocole supports the Airspeed data in a dedicated frame, but iNav needs to be adapted to be able to literally send the Airspeed data into the CRSF protocol... It needs to create a new frame (0A to be precise) containing the airspeed value and add this into the telemetry queue. This is exactly what the issue #11025 tries to create. Unfortunately it doesn't work that well and many bugs still need to be fixed.

sensei-hacker, 2025-12-18:
Someone tried to add airspeed in https://github.com/iNavFlight/inav/pull/11025 , but after I merged it we found out it broke other things. So I had to revert it.

If anyone wants to try to debug it, that would be cool.