iNavFlight/inav #9938: When I use angle mode during idle, the PID doesn't work

yxc407, 2024-04-12:
Why when I use angle mode during idle, the PID doesn't work and I can't use pitch, roll, and yaw for control? But when I use althold mode during idle, the PID does work?

b14ckyy, 2024-04-12:
Enable AIRMODE. that's exactly for that. 
[image]

yxc407, 2024-04-12:
> Enable AIRMODE. that's exactly for that. [...]

However, after enabling AIRMODE, even in angle mode at idle, there is still no PID correction and it cannot be controlled.

yxc407, 2024-04-12:
> Enable AIRMODE. that's exactly for that. [...]

This is in angle mode during idle.
[image]

This is in althold mode during idle.
[image]

b14ckyy, 2024-04-12:
Can you post a video please? What do you mean with IDLE? whent he quad is on the ground sitting the PID controls are supposed to be offline with no throttle. Airmode only kicks in when you throttle up to lift off. This is to prevent a freakout of the quad.

yxc407, 2024-04-12:
> Can you post a video please? What do you mean with IDLE? whent he quad is on the ground sitting the PID controls are supposed to be offline with no throttle. Airmode only kicks in when you throttle up to lift off. This is to prevent a freakout of the quad.

This is angle mode, motors idle power.
https://github.com/iNavFlight/inav/assets/143247595/c29026e7-1315-4ea5-93f4-699b78d0eb00
This is althold mode, motors idle power.
https://github.com/iNavFlight/inav/assets/143247595/637abc1f-4cce-4e61-bedb-1144e36dae20

yxc407, 2024-04-12:
> Can you post a video please? What do you mean with IDLE? whent he quad is on the ground sitting the PID controls are supposed to be offline with no throttle. Airmode only kicks in when you throttle up to lift off. This is to prevent a freakout of the quad.

I hope that for quadcopters, during the angle mode when the motors are idling, it can still allow control input and have the PIDs active, similar to how the althold mode works in inav. By idling, I mean when the throttle stick on the radio transmitter is at its lowest position after arming.

DzikuVx, 2024-04-12:
To unlock control THROTTLE has to be raised at least once above 20% mark. This is safety measurement to prevent windup before takeoff

closed 2024-04-12 by b14ckyy

yxc407, 2024-04-17:
> To unlock control THROTTLE has to be raised at least once above 20% mark. This is safety measurement to prevent windup before takeoff

Lastly, I'd like to ask, if I want to modify it to be like Betaflight where I can control the throttle below 20%, I would need to modify the code instead of adjusting the parameters?