iNavFlight/inav #9958: The default I2C address for baro sensor DPS310 in inav/src/main/target /common_hardware.c is wrong  (should be 0x77 and not 0x76) )

Thomasa66, 2024-04-18:
Regarding to the Infineon datasheet for the baro sensor DPS310 the default I2C address is 0x77 .

https://www.infineon.com/dgdl/Infineon-DPS310-DataSheet-v01_02-EN.pdf?fileId=5546d462576f34750157750826c42242
Described on page 12

In INAV : inav/src/main/target/common_hardware.c   the default I2C address for the DPS310 is set to 0x77 . (Thats wrong )

[image]

Hope that get fixed as soon as possible, or as alternative make it possible to switch/ set every sensor between default and substitute I2C address.

arturojs89, 2024-08-13:
Hi @Thomasa66 I have two barometer DSP310 and bmp280(internal), as soon as I connect the dsp310 the barometer doesn't works.

I assume is the issue you mentioned in the dsp310's datasheet it mentioned the I2C address is 0x77.

Have you found a way to set manually the baro address?

ShadowWalker8642, 2024-08-23:
its true i have board where i2c address for dps310 is 0x77 i had change it here temporarily but it worked after changing to 0x77