iNavFlight/inav #9959: New Target Request

enterobotica, 2024-04-18:
## Current Behavior
I can't find my FC board (HGLRCF405V2) as target in iNav configurator

## Desired Behavior
I want to install and use iNav in my 3,5" cinewhoop running on HGLRC SPECTOR F460

## Suggested Solution
I am willing to install iNav to test position hold and automated flight paths
Configuration:
Board: [HGLRC SPECTOR F460 (Stack)](https://www.hglrc.com/products/hglrc-specter-f460-stack-for-fpv-racing-drone-freestyle)
Default target name: HGLRCF405V2
Motors: Axisflying c206 2006 1960KV (6S)
RX Protocol: iBus
GPS: HGLRC m100 mini
Magnetometer: [LIS2MDL](https://robu.in/product/smartelex-lis2mdl-triple-axis-magnetometer/)
VTX: Analog (TBS Smartaudio 2.1)

## Who does this impact? Who is this for?
People using the mentioned hardware

## Additional context
Building this drone for smooth flying characteristics (capable to be controlled slow indoors and fast outdoor maneuvers) with return to home, position hold and GPS waypoint navigation for a reliable work drone that can also carry a full size GoPro.

I've just learnt that iNav does not require a magnetometer (although useful for accurate heading) to use for GPS enabled navigation features and I'm excited to try it out with my current hardware setup. I appreciate all development effort into the project.

b14ckyy, 2024-04-21:
For target requests, please follow the new hardware policy here: https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

enterobotica, 2024-04-23:
I have checked the new hardware policy, and my request does seem to be eligible for consideration. 

I have also emailed to dev team with the request code. 

Please suggest next steps 😊

b14ckyy, 2024-04-23:
Just wait for an answer from the core devs :)

sensei-hacker, 2024-04-23:
You might also choose to drop a note to HGLRC.  Normally the best way to get their product supported is for them to be involved.

https://hglrc.freshdesk.com/support/tickets/new

enterobotica, 2024-04-24:
I have got a response from HGLRC on a support ticket and have added them in the email to devs.

enterobotica, 2024-05-14:
Closing the request as the manufacturer has denied to support this request.

closed 2024-05-14 by enterobotica

CGPro856, 2024-08-28:
I have the same FC and the same issue. Was disappointed that BF didn't really support GPS and excited to flash to INAV but no support for the FC.

CGPro856, 2024-08-28:
@enterobotica, when you say manufacturer denied to support the request, what exactly does that mean? HGLRC doesn't want their F405V2 to be a target in iNav?

sensei-hacker, 2024-12-31:
FYI I anticipate this will be in INAV 8.1
https://github.com/iNavFlight/inav/pull/10417

sensei-hacker, 2025-01-02:
Merged for INAV 8.0, to be released soon.
https://github.com/iNavFlight/inav/pull/10417