iNavFlight/inav #9960: MATEK F405 WING V2 - Firmware Issue (INav 6.0 and up)

QuocSiPHU, 2024-04-19:
## Unable to connect to Matek F405 Wing V2 (Target MATEKF405SE)

## Steps to Reproduce

1. Open Inave 7.1, Plug in the FC in DFU mode (with the boot button)
2. Select Target MATEKF405SE (specified on the FC)
3. Load the firmware Online
4. Flash successfully, wait until led stop blinking, unplug the usb cable
5. Restart Inav, plug on the USB cable, click Connect (with the right port)
6. Inav open the serial port and after 10 sec, close the connection (communication failed)

## Expected behavior
Be able to connect to the board and setup the FC like another one.

## Suggested solution(s)
Fix the MATEKF405SE firmware for MATEKSYS F405 WING V2

## Additional context
The F405 Wing V2 got new sensors (ICM42688 and DSP310)
Apparently they are not recognized in either MATEKF405SE 6.0 nor 7.0
I've tried flashing with  MATEKF405 TE 6.0 and 7.0 as recommended on Mateksys website, however I still can't get connected

No dump available while I can't connect to the FC
---

- FC Board name and vendor: MATEKSYS F405 WING V2
- INAV version string: 7.0.0 MATEKF405SE - 2023-12-6 16:32 (stable)

sensei-hacker, 2024-04-19:
Is anything connected to the board, especially to uart 1 or UART 3?

QuocSiPHU, 2024-04-19:
Just only the board itself (No receiver, no servo, no GPS)
I've tried flashing with Betaflight (firmware 4.3.2 November 2021), got the connection, got the gyro and Baro working.
I connected my ELRS receiver and got channels responses on screen.
However no motor nor servo is working until now.

b14ckyy, 2024-04-19:
Did you enable full chip erase?
Can you please test if the same happens with inav 7.0 build from matek itself?

https://www.mateksys.com/?p=5159#tab-id-3

QuocSiPHU, 2024-04-19:
Hi b14ckyy,
Each flashing, I always do a full erase.
I just flash with link you provided (build 7.0 from Mateksys), and fortunately it work.
I can connect from Inav. I continue checking all the functions and will keep you aware.
Thanks

b14ckyy, 2024-04-19:
Can you please provide a detailed picture of the FC itself? 
Its a bit weird as the MATEKF405SE Target has not been changed over the last 5 Months and I am sure I have seen people using this FC with official INAV builds. 
@MATEKSYS has anything been changed on later revisions of this board?

QuocSiPHU, 2024-04-19:
> Did you enable full chip erase? Can you please test if the same happens with inav 7.0 build from matek itself?
> 
> https://www.mateksys.com/?p=5159#tab-id-3

Working hardware: Elrs Receiver, Servos, Led

not yet tested : Motor (have to be Ok)

not yet working :
GPS on UART4 (indoor testing, not getting satellites)
Buzzer (well wired) not getting any sound

QuocSiPHU, 2024-04-19:
> Can you please provide a detailed picture of the FC itself? Its a bit weird as the MATEKF405SE Target has not been changed over the last 5 Months and I am sure I have seen people using this FC with official INAV builds. @MATEKSYS has anything been changed on later revisions of this board?

[image]

QuocSiPHU, 2024-04-19:
I try powering the FC with rcv + servo + led (no USB to INAV)
The FC doesn't react to movement and the led is off (battery status)
The FC seems hanging!

b14ckyy, 2024-04-19:
Please also with the cover removed. 
Sorry for the extra effort but we would like to check if something is off with the used components.

QuocSiPHU, 2024-04-19:
[image]
[image]

QuocSiPHU, 2024-04-19:
I try to reflash twice but can't get the connection back! hic!

QuocSiPHU, 2024-04-19:
However If I reflash with Betaflight 4.3.2 (only this version), I will get the connection back and the gyro/baro are working normally!

OptimusTi, 2024-04-19:
[image]

Check these pins.

QuocSiPHU, 2024-04-19:
Those pins are intentionally connected in order to power S5 to S9 from the on board Bec.
It's indicated in the FC manual.
As I will not have more than 4 servos and it will not generate any over power consumption on the FC.

MATEKSYS, 2024-04-19:
> Can you please provide a detailed picture of the FC itself? Its a bit weird as the MATEKF405SE Target has not been changed over the last 5 Months and I am sure I have seen people using this FC with official INAV builds. @MATEKSYS has anything been changed on later revisions of this board?

nothing changed, just one PCB version since F405-WING-V2 release.

MATEKSYS, 2024-04-19:
Just try flashing INAV7.1.0 to F405-WING-V2, works normal.
Probably you had it flashed with wrong target.
post some screentshot of "firmware flasher" tab

"I've tried flashing with MATEKF405 TE 6.0 and 7.0 as recommended on Mateksys website, however I still can't get connected"
I am courious where did you see we recommended MATEKF405TE target for F405-WING-V2.

QuocSiPHU, 2024-04-19:
> Just try flashing INAV7.1.0 to F405-WING-V2, works normal. Probably you had it flashed with wrong target. post some screentshot of "firmware flasher" tab
>
> "I've tried flashing with MATEKF405 TE 6.0 [...]

Hi Mateksys, nice to read from you.
I'm using INAV 7.1.0 to flash the firmware [inav_7.0.0_MATEKF405SE]
from this link https://www.mateksys.com/Downloads/FW/inav_7.0.0_MATEKF405SE.hex

You talked about F405-WING-V2, but I have no idea where to get this firmware!

When power up the FC, I get the red led on and if the blue one is blinking, I can connect my FC to INAV.
If the blue led is off, I get "communication failed"
During the FC start up, I also have the green led blinking few time and then off.
I have no idea about the led signification.

Yesterday, I flashed twice but cannot get access from INAV.
Today, the board functioning but I'm not confident about its reliability.

If setup in a plane, the communication issue could be source of danger for people.
Thanks for your support.

QuocSiPHU, 2024-04-19:
> nothing changed, just one PCB version since F405-WING-V2 release.

[image]
[image]

QuocSiPHU, 2024-04-20:
> Just try flashing INAV7.1.0 to F405-WING-V2, works normal. Probably you had it flashed with wrong target. post some screentshot of "firmware flasher" tab
>
> "I've tried flashing with MATEKF405 TE 6.0 [...]

MATEKF405TE was a typing mistake from me!
Hereunder a screen shot showing some info one connected for your perusal.
[image]

sensei-hacker, 2024-04-20:
It might not hurt to take old toothbrush or something and clean up some of the extra solder bits around those pins - especially UART1 and UART3.  Then inspect with another close-up photo or a magnifying glass and strong light.

The soldering looks good on S1 and S9 - the shape shows both the pin and pad are wetted, with

QuocSiPHU, 2024-04-20:
> It might not hurt to take old toothbrush or something and clean up some of the extra solder bits around those pins - especially UART1 and UART3. Then inspect with another close-up photo or a magnifying glass and strong light. Just to be sure there aren't shorts from tiny solder whiskers and solder "dust" laying around on the board.
> 
> The soldering looks good on S1 and S9 - the shape shows both the pin and pad are wetted.
> 
> Because many of the other pins have enlarged balls of solder on them, and there are extra bits of solder in between, it's hard to know what might be shorted to what.

I did check the connection with a multimeter and there is no short on any pin.
However, your recommendation is right, and I'll clean up the board for removing unnecessary risk!
Thanks

QuocSiPHU, 2024-04-20:
> Just try flashing INAV7.1.0 to F405-WING-V2, works normal. Probably you had it flashed with wrong target. post some screentshot of "firmware flasher" tab
>
> "I've tried flashing with MATEKF405 TE 6.0 [...]

With the version 7.0.0 downloaded from your website, hopefully things keep working well.
Now I try to setup a buzzer but can't get any sound.
Could you help me please.
Thanks

QuocSiPHU, 2024-04-20:
The FC keep working!
Hope to definitively close this case. 
Thanks to all

closed 2024-04-20 by QuocSiPHU