iNavFlight/inav #9961: Remove OSD elements for unused features to avoid 'OSD orphans' in the layout

Painless360, 2024-04-19:
Many flight controllers now come with defaults set to CRSF as the default RC protocol, and also have some of the CRSF specific elements appearing in the OD layout. If you then change the RC protocol to SBUS, those CRSF elements are no longer shown on the OSD tab but will still be generated and displayed. 

This causes confusion as the user can see additional, unwanted elements in the OSD when viewing via the goggles, but can't access them to remove them in the OSD tab. To fix it, they need to identify the group that the OSD element comes from, enable it then remove it in the OSD Tab, before disabling it. Seems overly complex and confusing.

Could we have INAV automatically 'hide/disable' and hide OSD elements for things like CRSF automatically if CRSF isn't selected as the RC protocol?

Ideally, INAV only enabling the OSD elements for configured systems would make sense to avoid confusion.

This impacts any pilot setting up INAV without CRSF who doesn't remember to setup the OSD BEFORE they change the RC Protocol from CRSF to anything else.

Thanks for all the fab work on the project and for looking at this idea. Happy flying!

b14ckyy, 2024-04-21:
Actually this is not defined by the FC but by INAV itself. CRSF is the new default protocol since its now the most commonly used. 
But I agree this is  a good point for cleanup. 

Another example is Airspeed. When enabled in the OSD but then Airspeed is disabled in the Configuration, the element is hidden in the OSD config but still visible in the actual OSD.

Painless360, 2024-04-21:
Thanks b14ckyy, Airspeed is another good example. It would be nice to have INAV automatically disable and 'hide' any OSD icons from options that have not been enabled or had been, then tuned off. Thanks again!