iNavFlight/inav #9963: ESC stops working when enabling IRC Tramp

Chrilleguven, 2024-04-19:
Hi, when I turn on IRC Tramp on UART3 the ESC stops working.

I'm using a Mamba F405 MK2 board with a GPS and external Barometer. I have read that the speed of the MCU (looptime) can have an effect if you use I2C, so I have it set to max (800) but it does not change the outcome. 

## Expected behavior
ESC should work when enabling IRC Tramp

- FC Board name and vendor: Diatone Mamba F405 MK2 (FURYOSD Latest stable 7.1.0)
- INAV version string: 7.1.0

sensei-hacker, 2024-04-19:
Can we see a screenshot of your Ports tab, please?

Chrilleguven, 2024-04-19:
Hi, here is a screenshot! 
[image]

Jetrell, 2024-04-20:
I've noticed a similar issue with Tramp. But it doesn't stop the motor operating with Dshot.
What it does do, is very often leave the VTX in low power mode after arming.

This plane has a Matek 5.8G Tramp Vtx.. And it has always had the ability to switch VTx power levels in fight, from back in INAV 2.7 when it was built.. But of late, it has more flights that leave it stuck at 25mW. Than flights which I can increase the power to 500 or 800mW.. I only noticed it recently, because the video was much worse at than it was before. Which lead me see it stuck at zero on the OSD.. Although, 25mW is 1 and not zero. So I'm not sure about this.
[image]

For reference. This doesn't effect Smart Audio VTx power switching.

Chrilleguven, 2024-04-20:
@Jetrell But in my case it almost feels like the I2C is not working properly. The ESC is not even working as it should, only motors 1 and 4 are spinning. I cant read the setup in blheli configurator (blheli_s ESC). This occurs even if I have IRC Tramp, Baro or gps disabled. But with Betaflight everything works as expected.

Chrilleguven, 2024-04-20:
I think I found one issue regarding the motors! In the mixer tab, when timer outputs set on AUTO it messes up the timers: 
[image]

When setting the timer to MOTORS it read it correctly and the motors spin correctly: 
[image]

Chrilleguven, 2024-04-20:
It also fixed the issue with the IRC Tramp, but maybe something should be changed with the AUTO settings on the timers so others dont have to debug the same issues

closed 2024-04-20 by Chrilleguven