iNavFlight/inav #9964: FC freezes when using port setup DJI FPV VTX with (old) craft name hack

jreise-d, 2024-04-20:
We have found a bug that causes the FC to freeze after approx. 50 minutes of flight time. No more control is possible, the OSD freezes, there is no more output to the servos, regardless of which flight mode was currently active. No more LEDs flash on the FC, only a steady light.

We were lucky and were able to reproduce the error, localise it precisely with many tests and find the cause. It only occurs when DJI V1/V2 and DJI airunit is used without wtf.os with the UART setting _DJI FPV VTX_. If the _craft name_ in the OSD tab (e.g. from an old INAV config) is filled with the (old) hack string :WTDES and the OSD elements _System Messages_ and _craft name_ have also been placed on one of the four OSD screens and _Use craft name for messages_ is switched on, the freeze occurs after approx. 45 to 55 minutes, triggered by switching the flight mode.

We know that the craft name hack with the character string :WTDES is no longer necessary since INAV 4.0, but unfortunately it was copied from an old config. And unfortunately there are probably still leftovers in the code, which in this case lead to the FC freezing.

The error occurs reproducibly with several FCs, tested with Matek F405wing and H743wlite and with INAV 7.0 and 6.1.1. It was tested with different DJI airunits, also with different Crossfire RX, also the used GPS module was excluded.

DzikuVx, 2024-04-20:
@jreise-d please provide CLI `dump` as well

jreise-d, 2024-04-20:
here is a dump from our last test

[INAV_6.1.1_cli__WTDES_20240420_FreezeTest.txt](https://github.com/iNavFlight/inav/files/15049665/INAV_6.1.1_cli__WTDES_20240420_FreezeTest.txt)