iNavFlight/inav #9968: Motor keeps spinning, even after disarming

pintua65, 2024-04-22:
## Current Behavior
Very similar to closed issue #8795  on a fixed wing without prop I'm setting up. I set up calibration, receiver, ports, outputs, modes and I wanted to do a general test to gain confidence with "Modes" control.
I am in pure ACRO mode, all the "Tune" or "Safety" option are switched off.
After arming and raising the throttle over a value of 1300, the motor keep spinning at that speed (I don't know if it's the max speed) and I loose control over it. Disarming and Failsafe don't work. I have to unplug the batteries.
Throttle Sticks from 989->2012
Throttle Stick min value for motor to move: 1100

I searched a lot on forum and found some similar issues where there wasn't evidence of the cause. I tryed and change alternatively
nav_fw_min_thr  1200->1999
nav_fw_max_thr  1700->2000
nav_mc_hover_thr = 1300->2000
motor_pwm_rate = 16000->8000 (ESC default value is 16k)
without changing the result

I'll upload a log if needed

## Steps to Reproduce
1. Setup, calibrate and arm as per take off (plane is leveled horizontal)
2. Push throttle over a value around 1300 (in a scale of 1000-2000)
3. Try to pull the throttle to 0

## Expected behavior
I would expect the motor to stop when I pull the throttle to 0

## Additional context
[dump.txt](https://github.com/iNavFlight/inav/files/15066861/dump.txt)

INAV/SPEEDYBEEF405WING 7.1.0 Mar 27 2024 / 14:34:15 (59a6ee61)
GCC-10.3.1 20210824 (release)

Controller: Speedybee F405Wing
ESC: ZTW 30A
Motor: 2212 2450KV
INAV 7.1 and 6.1 also

sensei-hacker, 2024-04-22:
Can you please post a diff?  I want to examine and test your config on a different flight controller.

breadoven, 2024-04-23:
I assume the servos still work when the motor runs out of control ?

And what is the specific make of ESC used ? Could be you're using another standard ESC that doesn't like 400Hz, as was the case with #8795.

pintua65, 2024-04-24:
> Can you please post a diff? I want to examine and test your config on a different flight controller.

[diff INAV_7.1.0_cli_20240424_163137.txt](https://github.com/iNavFlight/inav/files/15096009/diff.INAV_7.1.0_cli_20240424_163137.txt)

Thank you

pintua65, 2024-04-24:
> I assume the servos still work when the motor runs out of control ?
> 
> Changing `motor_pwm_rate `only affects brushed motors, pmw rate for a Standard ESC is fixed at 400Hz. So it could be you're using another standard ESC that doesn't like 400Hz, as was the case with #8795. What is the specific make of ESC used ?

Thank you breadoven,

The ESC is a new one, here below an image and the manual that I read without finding any anomaly. I haven't done any motor calibration yet and I'm using a 3S Lipo pack
Might I change the 400kHz frequency? Do you suggest toward 800 or 200?
![Immagine 
[ZTW-Beatles-Series-ESC-User-Manual.pdf](https://github.com/iNavFlight/inav/files/15096127/ZTW-Beatles-Series-ESC-User-Manual.pdf)
WhatsApp 2024-04-24 ore 16 46 37_80395146](https://github.com/iNavFlight/inav/assets/167714148/4e5b1cb8-dc48-4e95-83c7-22611b9e42d2)

pintua65, 2024-04-24:
> Thank you breadoven,
>
> The ESC is a new one, here below an image and the manual that I read without finding any anomaly. I haven't done any motor calibration yet and I'm using a 3S Lipo pack Might I [...]

Update: I just tried with 800 and 200kHz and I get always the sameresult; attached a diff file for the 800kHz
[diff2 INAV_7.1.0_cli_20240424_163137.txt](https://github.com/iNavFlight/inav/files/15096572/diff2.INAV_7.1.0_cli_20240424_163137.txt)

breadoven, 2024-04-24:
Unfortunately the `motor_pwm_rate `is fixed at 400Hz if `motor_pwm_protocol `is set to STANDARD. `motor_pwm_rate` only affects brushed motors so changing it won't make any difference in your case. The only solution would be to use a different ESC.

pintua65, 2024-04-24:
I will change it and let you know, at least for future similar cases. I need a couple of days but in the meantime if any new idea/test arises let me know!
Thank you

pintua65, 2024-04-24:
I did a brief test logging the behaviour. After arming, I gently raised the throttle a bit and then put it to 0, then raised it again to cross the threshold. Then I tried to disarm without result as before
Thank you
[inav_data_log_2024-04-24_224108.zip](https://github.com/iNavFlight/inav/files/15100225/inav_data_log_2024-04-24_224108.zip)

closed 2024-04-25 by DzikuVx