iNavFlight/inav #9969: Mass Storage mode not working on SpeedyBee F405 Wing

tomaskir, 2024-04-23:
Greetings, I have issues getting `msc` to work with INAV.

FC and INAV:
- SpeedyBee F405 Wing
- SD card inserted in the FC
- INAV 7.1

Machines tested on:
- Desktop Machine running Linux (Manjar rolling)
- Laptop running Windows 10

Steps I am taking:
- connect USB cable to FC
- connect with Configurator (successful), and run `msc`

After the FC reboots, on the Windows laptop I simply get in Device Manager:
`Unknown USB Device (Device Descriptor Request Failed)`

On Linux, here are the `dmesg` logs:
```
[1419553.263467] usb 1-2: new full-speed USB device number 31 using xhci_hcd
[1419553.483507] usb 1-2: device descriptor read/64, error -71
[1419553.960463] usb 1-2: New USB device found, idVendor=0483, idProduct=5720, bcdDevice= 2.00
[1419553.960477] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1419553.960484] usb 1-2: Product: Mass Storage in FS Mode
[1419553.960491] usb 1-2: Manufacturer: STMicroelectronics
[1419553.960497] usb 1-2: SerialNumber: FFFFFFFEFFFF
[1419559.057962] usb 1-2: can't set config #1, error -110
```

Things I have tried:
- tested multiple different USB ports on the desktop
- 1 USB2 A port, 1 USB3 A port, 1 USB3 C port. All show same issue.
- tested with and without battery connected, same issue

After some discussion on Discord, also tested with:
```
sudo echo "Y" > /sys/module/usbcore/parameters/use_both_schemes
sudo echo "Y" > /sys/module/usbcore/parameters/old_scheme_first
```

This results in a slightly different log in `dmesg`, but the issue persists:
```
[1426011.862913] usb 1-2: new full-speed USB device number 47 using xhci_hcd
[1426011.889739] usb 1-2: Device not responding to setup address.
[1426012.123072] usb 1-2: Device not responding to setup address.
[1426012.329611] usb 1-2: device not accepting address 47, error -71
[1426012.519578] usb 1-2: new full-speed USB device number 48 using xhci_hcd
[1426012.705198] usb 1-2: New USB device found, idVendor=0483, idProduct=5720, bcdDevice= 2.00
[1426012.705210] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1426012.705216] usb 1-2: Product: Mass Storage in FS Mode
[1426012.705221] usb 1-2: Manufacturer: STMicroelectronics
[1426012.705225] usb 1-2: SerialNumber: FFFFFFFEFFFF
[1426017.747738] usb 1-2: can't set config #1, error -110
```

lienmeat, 2024-06-15:
This might not be related, but when using MSC, I found the size of the partition on the SD Card matters, and 4gb works, while 128gb does NOT for me (on Linux, Win, and Mac).  Try reformatting it to MBR FAT32 4gb, it might solve your issue, it might not.

Video I made about it:
[Youtube: Fix iNav Blackbox MSC Corruption With This Quick SD Card Solution!](https://youtu.be/oP3R7XIo8uE)

b14ckyy, 2025-04-07:
Fixed in 8.0

closed 2025-04-07 by b14ckyy