iNavFlight/inav #9974: Speedybee F405 Wing App - does not connect to inav-configurator software

TomComm27, 2024-04-25:
## Steps to Reproduce
1. Speedybee F405 Wing zusammengebaut und Batterie + ESC Kabel angelötet.
2. USB Verbinder mit FC und Laptop über USB-C verbunden
3. keine Fehler im Windows Geräte Manager aufgetaucht
4. INAV-Konfigurator 7.1 gestartet
5. Connect Button klicken -> Fehlermeldung: 
   2024-04-25 @ 06:14:57 -- Verbunden mit: COM4
   2024-04-25 @ 06:14:57 -- MSP-Verbindung erfolgreich geöffnet mit ID: 1
   2024-04-25 @ 06:15:07 -- Keine Konfiguration innerhalb von 10 Sekunden erhalten, Kommunikation fehlgeschlagen
   2024-04-25 @ 06:15:07 -- MSP-Verbindung erfolgreich geschlossen
6. Speedybee getrennt und mit gedrücktem Boot-Button wieder mit dem USB Kabel verbunden.
7. iNAV-Konfigurator zeigt beim Port "DFU" an
8. Unter dem Reiter Firmwareflasher werden beide Varianten probiert:
8.1 manuell heruntergeladene Firmware "inav_7.1.0_SPEEDYBEEF405WING.hex" ausgewählt
8.2 SPEEDYBEEF405WING ausgewählt und Firmware 7.1
9. immer mit "Ganzen Chip löschen" aktiviert 
10. "Flash Firmware" Taste gedrückt
11. Programmieren: ERFOLGREICH
12. Selber Fehler wiederfinden:
Connect button klicken -> Fehlermeldung: 
   2024-04-25 @ 06:14:57 -- Verbunden mit: COM4
   2024-04-25 @ 06:14:57 -- MSP-Verbindung erfolgreich geöffnet mit ID: 1
   2024-04-25 @ 06:15:07 -- Keine Konfiguration innerhalb von 10 Sekunden erhalten, Kommunikation fehlgeschlagen
   2024-04-25 @ 06:15:07 -- MSP-Verbindung erfolgreich geschlossen

## Expected behavior
It should connect to the FC and the properties/options should appear so that I can set the FC.

## Proposed solution(s)
If this is a bug, fix it please.

## Additional context
Unfortunately I can't get to the CLI tab.

- FC board name and manufacturer: Speedybee F405 Wing App
- INAV version string: 7.1 stable

Thidsa, 2024-04-27:
I have 2 of those SB 405W. Both are working here. Dont think its bug related

TomComm27, 2024-04-27:
What else could I try to find out what the problem is?

Thidsa, 2024-04-27:
Have you tried another computer, or connecting with your phone or pad with the speedybee app?

TomComm27, 2024-04-27:
I have tried it with my iPhone15 and an older Samsung Galaxy with the Speedybee APP. -> The Speedybee is found but an error message always appears when connecting.
I have also tried it with my laptop and desktop computer!

Regarding the app connection, unfortunately I cannot check UART6 to see if this is activated as I cannot establish a connection.

Übersetzt mit DeepL (https://www.deepl.com/app/?utm_source=ios&utm_medium=app&utm_campaign=share-translation)

[image]

Thidsa, 2024-04-27:
hm ok. Then something could be wrong with the flightcontroller. Have you tried to flash Betaflight or an older firmware?
Dont suppose I can come up with much more then

TomComm27, 2024-04-28:
I haven't tried Betaflight yet, I will definitely do that and give feedback!
What is exciting is that ArduPilot works!

Thidsa, 2024-04-28:
Well thats great. If you get Betaflight to work maybe a developer here can help you since they are quite alike

TomComm27, 2024-04-28:
I have now connected the Speedybee to Betaflight via DFU using a USB cable and installed the firmware file "betaflight_4.5.0_STM32F405.hex". The connection works but no "data" is displayed and I can't set almost anything:
[image]

[image]

[image]

Addition:
In the Betaflight Configuartor there is not exactly the right firmware in the FC firmware selection. I have now downloaded the best fitting one from github: "betaflight_4.5.0_STM32F405.hex"
Maybe this is the reason why I can connect but no data is loaded...

@Thidsa: Do you have INAV up on both your SP F405 Wing?

Thidsa, 2024-04-28:
Kind of great that you could connect. I just checked and on the speedybee page they say that there is no betaflight target for the wing board. So i've guess you flashed the target for 405 stack board for multirotors.
Anyway, maybe you could try now and flash inav again 'without' full chip erase, to see if you can retain the port settings you have now. Atleast thats what I would have done. You seem stuck anyway regarding how inav was yesterday.
If you cloud connect then your board maybe working and some dev here could help you get a custom target to make sure ports are like they should be.
@sensei-hacker what do you think?

Yes have latest inav on both sb 405 wing here.

Thidsa, 2024-04-28:
And if you then could connect, maybe at the same time load a dump or diff from me

Thidsa, 2024-04-28:
Ok. Sorry was a typo. Cloud should be Could.
I hoped the ports (uarts) would stick from betaflight to inav if you didnt full erase. So you could connect with usb or speedybee app. But maybe they dont anyway. I dont know. Hope some devs can help you then. Good luck

JoakimSelander, 2024-07-28:
I have the exact same problem. Can flash iNav but can't connect to it. Tried different computers.

90470, 2024-07-31:
Had the same problem. First thing connect the FC to the battery and then  put the FC into bluetooth mode, Blue light flashing. Make sure bluetooth is on in your phone or tablet, open the Speedybee app and it should connect. On my I phone, using WIFI I had to turn the FC on in WIFI mode White light flashing and then connect my phone to the Speedybee WIFI signal. After that go into the app and open. It did take awhile to connect. Hope this may help. Cheers

sensei-hacker, 2024-08-04:
Which target did you flash?
One common mistake is flashing the "Mini" target.  The Mini board is a quadcopter board, not compatible with the Wing or Wing Mini.

JoakimSelander, 2024-08-08:
> Which target did you flash? One common mistake is flashing the "Mini" target. The Mini board is a quadcopter board, not compatible with the Wing or Wing Mini.

Used the SPEEDYBEEF405WING target. Even tried the FW on Speedybee's page. Flashing works fine but can't connect to it afterwards.

sensei-hacker, 2024-08-08:
In that case, the next three things I would do:

Make sure you don't have any other programs running that will try to use the serial port. 
That includes the Arduino IDE, a 3D print slicer such as Cura, and ExpressLRS Configurator.

Disconnect or power off anything you have connected to UART1 or UART3.

Run the Zadig driver fixer if you're on Windows.

billyd60, 2025-06-29:
I had a lot of issues with the speedybee wing APP FC connecting to INAV and have isolated the problem at least for me, maybe this will help some people. In my case I was using a crossfire receiver on UART1 as instructed. However if the crossfire receiver is powered it will interfere with the USB connection and cause freezes or won't work at all in INAV. A simple quick fix is to remove the crossfire receiver from UART1 and connect it to UART2 and reset MSP on for UART1. But you must also power the FC separately for the USB to INAV connection to work. However, if you don't need to see receiver or mode functions operating with your radio, you can just disconnect the crossfire receiver and then you can connect via USB without a separate power source for the FC on UART1 or UART2. Of course you could use any other free UART 2 through 5 for the crossfire and achieve the same result.

TomComm27, 2025-07-01:
Hi guys, I am motivated to get the Speedybee F405 Wing App Flight Controller working again. Last year I received feedback from Speedybee support that they didn't have a solution either and that I would get my money back. Then I didn't want to try anything else.
Today I tried to install the firmware iNAV 8.0.1 which worked without errors, but I have the same problem again that it simply cannot establish a connection either via USB cable or via BLE etc.. *grrrr*
I took the plunge and ordered the exact same FC again this evening and it should arrive in 1-2 weeks. I'm really curious to see if it works with this one and if my model really had a hardware error or whatever (and maybe yours too) or if the same problem exists again.
I will report back to you in any case! 
P.S. @ billyd60: I have no sensor, no other USB device or anything attached to the FC or PC so that can't be causing the error!

OMGChase, 2025-07-07:
> Hi guys, I am motivated to get the Speedybee F405 Wing App Flight Controller working again. Last year I received feedback from Speedybee support that they didn't have a solution either and that I [...]

I honestly am starting to question wether it's INAV or this flight controller. I was doing a lot of tests this weekend, and it seems that it sometimes just isn't behaving as a normal FC should behave. I had INAV 6.0.1 on it (the only one that seems to work) and after trying to save settings in INAV, it completely bricked the FC and had to reflesh. And even reflashing doesn't work 100% of the time. I would need to flash it multiple times until it eventually sticks. I have also contacted Speedybee for a refund. Ardupilot on the other hand does flash properly, but I've had issues with servos and the esc not working together. I'm not sure if I want to get another one of these flight controllers, it's been nothing but uphill.

TomComm27, 2025-07-16:
You won't believe it! My new speedybee f405 wing app arrived today and I installed the inav 8.0.1 firmware on it immediately after assembly without any problems. 
After unplugging and reconnecting to the computer, the big moment came to see if I could connect to the inav configurator and the FC!
JAAAA it worked without any problems. Guys, I'm telling you, that was a good feeling because I was really getting desperate.

So colleagues, if you have tried something similar to me and have not found a connection. It's up to the FC! Complains about the thing and orders a new one!

This solves the problem. 
Fg Thomas

closed 2025-07-16 by TomComm27