iNavFlight/inav #9975: Unable to arm "throttle not low" even when set

IonescuSergiuMihai, 2024-04-26:
I have a problem with the inav settings with "throttle not low"
[image]
this thirst min throttle
[image]
it wants max throttle

sensei-hacker, 2024-04-26:
Can you please attach your diff.

With default settings, throttle must be below 1100. With bidirectional motors turned on, it must be close to 1500.

IonescuSergiuMihai, 2024-04-26:
# diff

# version
# INAV/SPEEDYBEEF405WING 7.0.0 Dec  5 2023 / 11:06:36 (895a4f31)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# resources

# Timer overrides

# Outputs [servo]
servo 1 1000 2000 1500 -100

# safehome

# features
feature GPS
feature REVERSIBLE_MOTORS
feature PWM_OUTPUT_ENABLE
feature FW_AUTOTRIM

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# Receiver: Channel map

# Ports

# LEDs

# LED color

# LED mode_color

# Modes [aux]
aux 0 0 0 1400 2100
aux 1 2 3 1800 2100
aux 2 10 1 1650 2100
aux 3 3 2 1300 1700
aux 4 54 2 1650 2100

# Adjustments [adjrange]

# Receiver rxrange

# temp_sensor

# Mission Control Waypoints [wp]
#wp 0 invalid

# OSD [osd_layout]
osd_layout 0 0 27 0 V
osd_layout 0 1 11 14 V
osd_layout 0 2 0 0 V
osd_layout 0 3 8 6 V
osd_layout 0 4 8 6 V
osd_layout 0 7 14 1 V
osd_layout 0 9 0 2 V
osd_layout 0 11 0 3 V
osd_layout 0 12 0 4 V
osd_layout 0 14 0 6 V
osd_layout 0 23 1 1 V
osd_layout 0 28 24 5 V
osd_layout 0 29 2 12 H
osd_layout 0 30 2 13 V
osd_layout 0 33 0 5 V
osd_layout 0 34 11 12 V
osd_layout 0 38 16 14 V
osd_layout 0 48 24 3 V
osd_layout 0 49 25 2 V
osd_layout 0 87 25 4 V
osd_layout 0 104 10 3 H
osd_layout 0 119 8 4 H
osd_layout 0 125 1 5 H
osd_layout 0 126 1 6 H
osd_layout 0 129 10 0 V
osd_layout 0 138 0 6 H
osd_layout 0 144 0 5 H

# Programming: logic

# Programming: global variables

# Programming: PID controllers

# master
set gyro_main_lpf_hz = 25
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set gyro_zero_x = 1
set gyro_zero_y = -16
set gyro_zero_z = 10
set ins_gravity_cmss =  984.643
set acc_hardware = ICM42605
set acczero_y = -8
set acczero_z = 34
set accgain_x = 4101
set accgain_y = 4092
set accgain_z = 4103
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = SPL06
set min_check = 1050
set serialrx_provider = IBUS
set serialrx_inverted = ON
set motor_pwm_protocol = STANDARD
set failsafe_procedure = RTH
set small_angle = 180
set applied_defaults = 4
set gps_provider = UBLOX7
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_ublox_use_beidou = ON
set airmode_type = STICK_CENTER_ONCE
set nav_wp_radius = 5000
set nav_wp_max_safe_distance = 500
set nav_rth_allow_landing = FS_ONLY
set nav_rth_altitude = 5000
set nav_fw_bank_angle = 45
set nav_fw_control_smoothness = 2
set nav_fw_launch_max_angle = 75
set nav_fw_launch_motor_delay = 100
set nav_fw_launch_max_altitude = 5000
set nav_fw_launch_climb_angle = 25
set nav_fw_allow_manual_thr_increase = ON
set tz_automatic_dst = EU

# mixer_profile
mixer_profile 1

set platform_type = AIRPLANE
set model_preview_type = 8
set motorstop_on_low = ON

# Mixer: motor mixer

mmix reset

mmix 0  1.000  0.000  0.000  1.000
mmix 1  1.000  0.000  0.000  1.000

# Mixer: servo mixer
smix reset

smix 0 1 0 50 0 -1
smix 1 1 1 50 0 -1
smix 2 2 0 -50 0 -1
smix 3 2 1 50 0 -1

# profile
profile 3

# battery_profile
battery_profile 1

set throttle_idle =  5.000

# save configuration
save

# end the command batch
batch end

sensei-hacker, 2024-04-26:
Yep, you have reversible motors on. So zero throttle is in the middle.
1,000 would be full speed backward.

IonescuSergiuMihai, 2024-04-26:
[image]

IonescuSergiuMihai, 2024-04-26:
that's kind of the problem

IonescuSergiuMihai, 2024-04-26:
# diff

# version
# INAV/SPEEDYBEEF405WING 7.0.0 Dec  5 2023 / 11:06:36 (895a4f31)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# resources

# Timer overrides

# Outputs [servo]
servo 1 1000 2000 1500 -100

# safehome

# features
feature GPS
feature PWM_OUTPUT_ENABLE
feature FW_AUTOTRIM

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# Receiver: Channel map

# Ports

# LEDs

# LED color

# LED mode_color

# Modes [aux]
aux 0 0 0 1400 2100
aux 1 2 3 1800 2100
aux 2 10 1 1650 2100
aux 3 3 2 1300 1700
aux 4 54 2 1650 2100

# Adjustments [adjrange]

# Receiver rxrange

# temp_sensor

# Mission Control Waypoints [wp]
#wp 0 invalid

# OSD [osd_layout]
osd_layout 0 0 27 0 V
osd_layout 0 1 11 14 V
osd_layout 0 2 0 0 V
osd_layout 0 3 8 6 V
osd_layout 0 4 8 6 V
osd_layout 0 7 14 1 V
osd_layout 0 9 0 2 V
osd_layout 0 11 0 3 V
osd_layout 0 12 0 4 V
osd_layout 0 14 0 6 V
osd_layout 0 23 1 1 V
osd_layout 0 28 24 5 V
osd_layout 0 29 2 12 H
osd_layout 0 30 2 13 V
osd_layout 0 33 0 5 V
osd_layout 0 34 11 12 V
osd_layout 0 38 16 14 V
osd_layout 0 48 24 3 V
osd_layout 0 49 25 2 V
osd_layout 0 87 25 4 V
osd_layout 0 104 10 3 H
osd_layout 0 119 8 4 H
osd_layout 0 125 1 5 H
osd_layout 0 126 1 6 H
osd_layout 0 129 10 0 V
osd_layout 0 138 0 6 H
osd_layout 0 144 0 5 H

# Programming: logic

# Programming: global variables

# Programming: PID controllers

# master
set gyro_main_lpf_hz = 25
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set gyro_zero_y = -17
set gyro_zero_z = 12
set ins_gravity_cmss =  985.629
set acc_hardware = ICM42605
set acczero_y = -8
set acczero_z = 34
set accgain_x = 4101
set accgain_y = 4092
set accgain_z = 4103
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = SPL06
set min_check = 1050
set serialrx_provider = IBUS
set serialrx_inverted = ON
set motor_pwm_protocol = STANDARD
set failsafe_procedure = RTH
set small_angle = 180
set applied_defaults = 4
set gps_provider = UBLOX7
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_ublox_use_beidou = ON
set airmode_type = STICK_CENTER_ONCE
set nav_wp_radius = 5000
set nav_wp_max_safe_distance = 500
set nav_rth_allow_landing = FS_ONLY
set nav_rth_altitude = 5000
set nav_fw_bank_angle = 45
set nav_fw_control_smoothness = 2
set nav_fw_launch_max_angle = 75
set nav_fw_launch_motor_delay = 100
set nav_fw_launch_max_altitude = 5000
set nav_fw_launch_climb_angle = 25
set nav_fw_allow_manual_thr_increase = ON
set tz_automatic_dst = EU

# mixer_profile
mixer_profile 1

set platform_type = AIRPLANE
set model_preview_type = 8
set motorstop_on_low = ON

# Mixer: motor mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.300
mmix 1  1.000  0.000  0.000 -0.300

# Mixer: servo mixer
smix reset

smix 0 1 0 50 0 -1
smix 1 1 1 50 0 -1
smix 2 2 0 -50 0 -1
smix 3 2 1 50 0 -1

# profile
profile 3

# battery_profile
battery_profile 1

set throttle_idle =  5.000
set nav_fw_min_thr = 1100
set nav_fw_launch_idle_thr = 1600

# save configuration
save

# end the command batch
batch end

sensei-hacker, 2024-04-26:
Let's clarify - are you needing to run your motors backwards?

IonescuSergiuMihai, 2024-04-26:
and how do I do this at ESC, and I haven't disabled it from INAV?
[image]

sensei-hacker, 2024-04-26:
It looks like it IS disabled in INAV now, so that's great.  Depending on which ESC you have, the easiest way to configure it might be esc-configurator.com 
You don't have to download anything from there - the web site connects to your ESC, through the USB cable plugged into your flight controller.

IonescuSergiuMihai, 2024-04-26:
But surely there isn't something from INAV that can solve the throttle?
I don't really think there would be problems with ESC, especially since I did a test a long time ago and it went well.