iNavFlight/inav #9977: BLADE F7 45A AIO

meor97, 2024-04-27:
## Current Behavior

## Desired Behavior
New target for Inav

## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

sensei-hacker, 2024-04-27:
The manufacturer's product page indicates the RUSH_BLADE_F7 target is the correct target for that board.
Are you having some trouble with it?

sensei-hacker, 2024-04-27:
The best way to get new hardware supported is to ask the manufacturer to please get in touch and provide us the necessary information. The best and normal process is that the manufacturer provides us the information about their design so we can support it.

https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

meor97, 2024-04-27:
It doesnt work. I flash Inav using blade f7 all the sensors are off

sensei-hacker, 2024-04-27:
Yes, the manufacturer would need to provide information for THIS board.

stewwy, 2024-11-19:
Any movement on this?