iNavFlight/inav #9982: Update external logging

Hex-Davis, 2024-04-27:
## Current Behavior

The external logging seems to be outdated. The Openlog Artemis is able to log at a baud of 500,000
## Desired Behavior

support for the Openlog Artemis in general, support in the ports tab for higher peripheral baud rates, and some updates to the Black box setup guide reflecting the changes.
## Suggested Solution

verify  higher peripheral baud rates are possible and create an option in the peripherals under the ports tab for the higher values
## Who does this impact? Who is this for?

anyone using a FC without full Blackbox support ie. H7, F7...
## Additional context

muhammedoney, 2025-02-24:
I was trying to get logs to SD card with openlog but there is a problem. It writes some initial data correctly, then writes some corrupted data while writing the logs of sensor and stick inputs. It seems like a baudrate confusion but I haven't found the right setting yet. I need your help, please see the documents attached

[LOG00044.TXT](https://github.com/user-attachments/files/18941633/LOG00044.TXT)
[config.txt](https://github.com/user-attachments/files/18941635/config.txt)

muhammedoney, 2025-02-24:
thank you, i guess i was halusinating because it took so much time. and i was expecting whole notations would be the ASCI but appearently no