iNavFlight/inav #9984: Only one servo works  f405 v2 wing

InzynierMamon, 2024-04-28:
Working on a fixed wing (full tail, flaps), matek f405-wing v2, with spectrum ESC Tx and Rx.  Rx is hooked into serial receiver--only the s3 servo moves when moving Tx sticks, motor will not turn using radio, it turns when adding throttle  in inav.  channel map reflects true stick movement for all like there is no problem .  Servos test OK. What could be the issue? Thanks in advance.

sensei-hacker, 2024-04-28:
Sounds like mixer issues. Can you please attach a screenshot of your mixer.

Also double check that the motor and servos are actually attached to the outputs shown in your mixer.

InzynierMamon, 2024-04-29:
On closer look s3 (elevator) and s4 (left aileron) are working 

S1 ESC Vx pin removed, servos have Vx connected

Receiver on Tx2

More and more I'm thinking it has to be the solder points (although I think I soldered well) --- what would be the best way to check--hook up voltmeter and move the radio sticks?

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

sensei-hacker, 2024-04-29:
The first thing I would check with a multimeter is remove power and check continuity between each pair of adjacent pins. 

If you find continuity between pins, also check continuity to ground.

MATEKSYS, 2024-05-03:
[image]

InzynierMamon, 2024-05-04:
Vx2 to Vx pad bridge created--all servos now working -- thanks for the pointers. 

Now just S1 ESC/motor doesn't respond to throttle stick inputs on Tx(everything in iNAV indicates it should), have Vx disconnected in the ESC S1 pins.

InzynierMamon, 2024-05-05:
I'm unable to calibrate ESC -- not allowed to turn on motor test mode and motor value at 1460.  

[image]

sensei-hacker, 2024-05-17:
> I'm unable to calibrate ESC -- not allowed to turn on motor test mode and motor value at 1460.

Yeah, you can't enable the motor with the throttle already halfway half. Drop the throttle stick to zero and turn off bidirectional motors if you don't actually need to run your motor backwards.  If you're building a rover, which does need to be able to go backwards, that's a different procedure.

closed 2024-05-17 by sensei-hacker