iNavFlight/inav #9985: Using rangefinder in INAV7.1 causing the quad floating up and down

Minderring, 2024-04-28:
## Current Behavior
The optic flow & lidar sensor MTF-01 work well in INAV6.1 and INAV7.0 as the video: https://www.youtube.com/watch?v=bEKm-PGRnks
But when I upgraded the firmware to INAV7.1, the quad floated up and down: https://www.youtube.com/shorts/A4Gc1CagAus
Same hardware and configuration, but different results.
To find the reason. I tried the ALTHold mode using baro and it worked well, then I added the rangefinder and in Surface Mode and ALTHold mode, the problem shown in the video has occurred.
[image]

## Steps to Reproduce
1. Enable rangefinder and check it works well
2. Turn to ANGLE ALTHOLD and SURFACE mode together
3. Arm and take off

## Expected behavior
I tested it in INAV6.1 and INAV7.0 and it flied stably. I consider it should be the same as previous. 

## Suggested solution(s)
If there is any change about the fusion of baro and rangefinder between INAV7.0 and INAV7.1 which might cause the problem.

## Additional context
# version
# INAV/MicoAir405v2 7.1.0 Apr 20 2024 / 17:47:36 (aa854365)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# resources

# Timer overrides
timer_output_mode 0 AUTO
timer_output_mode 1 MOTORS
timer_output_mode 2 MOTORS
timer_output_mode 3 AUTO
timer_output_mode 4 AUTO
timer_output_mode 5 AUTO
timer_output_mode 6 AUTO
timer_output_mode 7 AUTO
timer_output_mode 8 AUTO
timer_output_mode 9 AUTO
timer_output_mode 10 AUTO
timer_output_mode 11 AUTO
timer_output_mode 12 AUTO
timer_output_mode 13 AUTO

# Outputs [servo]
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
safehome 6 0 0 0
[... 1612 lines cut ...]
set nav_fw_cruise_thr = 1400
set nav_fw_min_thr = 1200
set nav_fw_max_thr = 1700
set nav_fw_pitch2thr = 10
set nav_fw_launch_thr = 1700
set nav_fw_launch_idle_thr = 1000
set limit_cont_current = 0
set limit_burst_current = 0
set limit_burst_current_time = 0
set limit_burst_current_falldown_time = 0
set limit_cont_power = 0
set limit_burst_power = 0
set limit_burst_power_time = 0
set limit_burst_power_falldown_time = 0

# save configuration
save

# end the command batch
batch end

---
# version
# INAV/MicoAir405v2 7.1.0 Apr 20 2024 / 17:47:36 (aa854365)
# GCC-10.3.1 20210824 (release)

DzikuVx, 2024-04-29:
This is caused by the fact that lidar is not able to get stable readings. Whenever lidar becomes unstable INAV executes descend to regain altitude readings

Minderring, 2024-04-29:
Thank you! But it performs normally on INAV 6.1 and 7.0, abnormally on INAV 7.1. 
The height is also within the measurement range. I check the data of the rangefinder in real-time through telemetry and is correct.

breadoven, 2024-04-29:
It would be useful if you could provide a Log and also a Diff file (easier to check).

Minderring, 2024-04-30:
Okay, I will test it and save the log later.