iNavFlight/inav #9986: SITL: inable to select serial receiver

RomanLut, 2024-04-28:
It is not possible to select any "SERIAL" receiver in SITL.
Tested with current Master branch.

I am sure it was possible to select SERIAL SBUS or CRSF before:
https://github.com/iNavFlight/inav/pull/9365

stronnag, 2024-04-29:
This is probably a configurator issue, not a firmware issue.

IBUS works 
```
inav_8.0.0_SITL --path ~/sitl-eeproms/ibus-eeprom.bin --serialuart=3 --serialport /dev/ttyUSB0 --baudrate 115200
```
CRSF works
```
inav_8.0.0_SITL --path ~/sitl-eeproms/crsf-eeprom.bin --serialuart=3 --serialport /dev/ttyUSB0 --baudrate 420000
```

From the configurator, spawning `ser2tcp` is indeed broken as a result of the Electron merge; however launching `ser2tcp` from a terminal also works.

In all cases, stick movement is observed in the configurator.

Not a fiirmware issue, at least here.

stronnag, 2024-04-29:
Please reopen if you can demonstrate a use case that indicates a firmware problem.

closed 2024-04-29 by stronnag

RomanLut, 2024-04-29:
Seems to be a configurator issue, but I can be sure when other bugs are fixed (at least not working cli with SITL).

stronnag, 2024-04-29:
> Seems to be a configurator issue, but I can be sure when other bugs are fixed (at least not working cli with SITL).

Either a Configurator issue or OS specific, CLI with SITL works OK here.
```
$ cliterm tcp://localhost:5760
opening  tcp://localhost:5760 ...

Entering CLI Mode, type 'exit' to return, or 'help'

# status
INAV/SITL 8.0.0 Apr 27 2024 / 16:49:35 (e92337a5)
GCC-13.2.1 20231014
System Uptime: 31 seconds
Current Time: 2024-04-29T12:39:01.000+01:00
Voltage: 16.79V (4S battery - OK)
CPU Clock=500MHz, GYRO=FAKE, ACC=FAKE, BARO=FAKE, PITOT=VIRTUAL
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
System load: 0, cycle time: 1001, PID rate: 999, RX rate: 19, System rate: 10
Arming disabled flags: CMS
OSD: not enabled
VTX: not detected

# 
Leaving CLI mode, unsaved changes lost.

Rebooting
```