iNavFlight/inav #9987: VSpd telemetry error accumulates in flight

dingrgh, 2024-04-28:
## Current Behavior
VSpd telemetry parameter accumulates an error with a offset of up to +2m/s after about 30 minutes in the air. At the start of the flight, there is zero error. At the end of the flight, with the aircraft sitting on the ground, telemetry shows it is still climbing. This error is also noticeable in flight if the vario tones are enabled (i.e. a climb indicated when the glider is actually descending).
[image]

## Steps to Reproduce
1. Arm. Launch and fly.
2. After about, say, 30 minutes in the air, land.
3. Check VSpd telemetry with the aircraft sitting on the ground to observe the offset.
4. Confirm that the Alt telemetry figure is correct  (i.e. not changing).

## Expected behavior
At the end of the flight, the VSPD telemetry should show zero m/s when sitting stationary on the ground.

## Suggested solution(s)
The error was probably introduced in code changes for barometric and GPS alt.

## Additional context
This VSpd error only started after I updated from iNav 6.0 to iNav 7.1.0.
iNav Dump: (https://pastebin.com/7ZSPq3p6)

---

- FC Board name and vendor: Matek F405-WSE
- INAV version string: INAV/MATEKF405SE 7.1.0 Mar 27 2024 / 13:52:59 (59a6ee61)

dingrgh, 2024-05-02:
I did more flights with iNav 7.1.0 to confirm that the problem was repeatable. I even tried setting CLI parameter inav_w_z_gps_v to zero (default is 0.1) to remove all dependence on GPS vertical speed but it did not fix the problem.

I downgraded the firmware to iNav 7.0.0 and can confirm that the problem went away. I can only conclude that some change to the Vertical Speed code from 7.0.0 to 7.1.0 is causing the problem.

Summary: Rolling back to iNav 7.0.0 is a workaround for this issue.

Frank0587, 2024-05-12:
Hey, 
I had similar problems. The Baro was no longer used for altitude and vario. The default setting for “inav_use_gps_no_baro” has been changed from OFF to ON. After the CLI command “set inav_use_gps_no_baro = OFF” it works perfectly again. I hope this will solve your problem too.

dingrgh, 2024-05-14:
> Hey, I had similar problems. The Baro was no longer used for altitude and vario. The default setting for “inav_use_gps_no_baro” has been changed from OFF to ON. After the CLI command “set inav_use_gps_no_baro = OFF” it works perfectly again. I hope this will solve your problem too.

I will try this and report back in a week or so...

dingrgh, 2024-05-15:
I wonder if the issue is related to #10040.

sensei-hacker, 2024-05-15:
> The VSpd issue may be related to #10040.

I created #100400 based partly on reading this. Suggest turning off inav_use_gps_no_baro.

dingrgh, 2024-05-17:
I have done several flights (iNav 7.1.1) with  “inav_use_gps_no_baro” set to OFF. Unfortunately, the VSpd residual offset problem is still present.

So the problem is not likely to be related to #10040 but is a result of something that was changed in the Z velocity position estimator code between iNav 7.0.0 and iNav 7.1.x.

dingrgh, 2024-05-26:
Update: Workaround found...

Version: iNav 7.1.1

Things I tried that didn't solve the problem:
- Setting "inav_use_gps_no_baro" to OFF.
- Setting "inav_w_z_gps_v" to zero (default was 0.1) in an attempt to remove all GPS vertical speed weighting
- Setting "inav_w_z_baro_p" to 2.0 (default was 0.35) in an attempt to increase weighting of baro altitude.

What worked finally:
- I kept the above three changes and also changed "inav_w_z_gps_p" to zero (default was 0.2) to remove all GPS altitude weight. The VSpd telemetry now shows zero after landing (other than the expected small blips plus or minus 0.1m/s due to baro noise) while sitting stationary on the ground.

As mentioned in a previous post, rolling back to iNav 7.0.0 (without the changes above) also works around the problem.

Clearly, the problem introduced by code changes in iNav 7.1.0 will have to be fixed as I suspect it will affect other navigation functions in some annoying, hard-to-pinpoint way.

silicongarage, 2024-07-10:
Problem is still not fixed in latest iNav 7.1.2

VSpd show +2 ft/sec accumulated upon landing after ~30min flying.  
This is what I've tried.
- set `inav_use_gps_no_baro = OFF`
- set `inav_w_z_gps_v = 0`

I have not tried `inav_w_z_gps_p = 0` as I want the GPS to correct any altitude drift of the Baro

HW:  Speedybee F405 Wing
FW:  iNav 7.1.2

dingrgh, 2025-04-18:
I just updated to iNav 8.0.1 and did a few test flights. The residual Vspd problem still exists in iNav 8.0.1.

You don't have to actually do a flight to see the problem. It will manifest itself on the ground. You can see this by:

1. Power up TX and iNav FC.
2. Wait to acquire the requisite number of GNSS satellites.
3. Arm.
4. Wait and watch the Vspd telemetry on the TX. After a few minutes, the Vspd value will start to accumulate even though the aircraft is stationary on the ground.

The workaround is the same... 
set inav_w_z_gps_p = 0.0

This works for me as I am happy to rely solely on Baro altitude and do not need GPS augmentation for altitude. But it is not ideal for most other flyers.