iNavFlight/inav #9994: GPS HGLRC M100 5883 is taking a long time to capture and fix satellites on inav 7.1

Marvin1972, 2024-04-30:
I'm having problems using the HGLRC M100 5883 GPS. Using the inav 7.1 version, it's taking more than 20 minutes to capture satellites and fix 3d. The GPS is well isolated from interference and I shielded the cable with copper tape. Using the same GPS model in version 6.1 of INAV, it captures satellites normally and in a reduced time. Please check this bug.

## Additional context

https://pastebin.com/evzbGcnq

# version
# INAV/MATEKF405SE 7.1.0 Mar 27 2024 / 13:52:59 (59a6ee61)
# GCC-10.3.1 20210824 (release)

sensei-hacker, 2024-04-30:
New GPS units take a minimum of 12.5 minutes outdoors to download the emperides and almanac.

12 1/2 minutes is the fastest possible, since that's the rate the satellites send them. Interference from a VTX or a telemetry receiver will make it take longer.

You will probably want to enable Galileo as well for best performance.

If you'd like, I can provide further tips if you want to attach a photo of how it's currently mounted and what is mounted near it.

rx9abc, 2024-05-12:
> I'm having problems using the HGLRC M100 5883 GPS. Using the inav 7.1 version, it's taking more than 20 minutes to capture satellites and fix 3d. The GPS is well isolated from interference and I shielded the cable with copper tape. Using the same GPS model in version 6.1 of INAV, it captures satellites normally and in a reduced time. Please check this bug.
> 
> ## Additional context
> https://pastebin.com/evzbGcnq
> 
> # version
> # INAV/MATEKF405SE 7.1.0 Mar 27 2024 / 13:52:59 ([59a6ee6](https://github.com/iNavFlight/inav/commit/59a6ee6118f4eb06bbed50c0f0520d3cb7fbad30))
> # GCC-10.3.1 20210824 (release)

I have exactly the same problem

sensei-hacker, 2024-05-12:
@rx9abc  if you'd like some assistance z please provide a copy of your diff and a photo of your GPS mounting. Also be sure to do all testing outdoors. Traditionally, GPS isn't supposed to work indoors - but the newest units are so good they sometimes pull it off even indoors.

rx9abc, 2024-05-13:
@sensei-hacker Yeah, I run all my tests outside. The GPS works like a charm with INAV 6, but as soon as I flash INAV 7, the GPS conks out. Once I reflash INAV 6, the GPS is up and running again.
[image]
[INAV_7.1.1_cli_20240513_225602.txt](https://github.com/iNavFlight/inav/files/15298363/INAV_7.1.1_cli_20240513_225602.txt)

sensei-hacker, 2024-05-13:
@rx9abc  Can you please post a screenshot of your GPS tab.  I want to see the "messages".and "errors" numbers, as well as confirming the config (though I did read your diff).

FYI if you want to use the magnetometer now or in the future, the two worst things to have near the mag are a) the main battery wires and b) an active buzzer.

rx9abc, 2024-05-13:
@sensei-hacker 
[image]

During flights with INAV6, I had no issues at all with picking up the GPS signal, even with the power cable hooked up. When I'm testing INAV7, I just skip the power battery altogether.
The tBeacon doesn't work in during flight. It only activates after crash.

[image]

sensei-hacker, 2024-05-13:
Ah, you also have another mag separate from your GPS, a long with the 5883 built into the HGLRC.  Cool.  _GPS_ near the power wires and buzzer should be okay.

Since the messages count is high and errors is low, that tells us that communication between the GPS and the FC is okay. The GPS itself just isn't getting a fix.  That's odd, if it has had more than 12 minutes outdoors to download the almanac and ephemerides.

Since we're now in strange territory, have you tried turning off Galileo just to see what happens?  It shouldn't be necessary, but this _should_ be working exactly as you have it, from everything I can see.

Also I don't suppose you used U-Center to change any GPS settings?  I ask because the settings set by INAV assume that other settings are at the default.

rx9abc, 2024-05-13:
Yeah, looks like it's time to get connected with the U-center. I'm not seeing any other way to troubleshoot this right now. I'll disable Galileo for the next test run.

sensei-hacker, 2024-05-13:
If you haven't _already_ made any changes via U-Center, please hold off for now.  Also I forgot to ask - which GPS unit do you have?  You're in a thread titled "GPS HGLRC M100 5883", but I'm starting to suspect that's not what you have.  I'm starting to think maybe you have a Neo 6 or similar.

rx9abc, 2024-05-14:
Exactly, it`s my mistake, I didn't look at the topic title. I use GPS from RadioLink model TS 100.