iNavFlight/inav #9998: No servo outputs on a Matek F405-SE quad setup in iNav 7.1.0

mrduncan626, 2024-04-30:
I have a quadcopter setup with a Matek F405-SE and iNav 6.1.0.  Setup with 4 motor outputs (S1, S2, S3, S4) and configured with 3 servo outputs (S7, S8, S9).  All the servo outputs work as program on the radio (RadioMaster TX16s)

After updating to iNav 7.1.0, none of the servo outputs are functional.
   a. All 3 servo channels are functional as indicated in the Configurator Receiver tab and the Outputs tab.
   b. Motor and Servo outputs are enabled.
 
Re-installed iNav 6.1.0 and all the servo outputs are functional again.

mrduncan626, 2024-04-30:
[image]

mrduncan626, 2024-04-30:
[image]

sensei-hacker, 2024-04-30:
You said you have the servos wires to S7 - S9.
Your mixer is configured for servos on S5 - S7.

Those need to match, of course.
You may want to set group 8 (timer 8) to motors mode, which will push your servos up to S7-S9.

closed 2024-04-30 by sensei-hacker