iNavFlight/inav #10000: OSD for analog shows wrong stuff  in INAV 7.1.1

wterreb, 2024-05-02:
## Current Behavior
I try to to set up OSD an my analog quad in 7.1.1 (self compiled binaries).
The OSD elements shows up correctly in the OSD tab of the configurator.
But when you look in the goggles it is all wrong

## Steps to Reproduce
1.  Download INAV for 7.1.1 from Github
2.  Build for FLYWOOF745NANO and flash to quad
3.  Look what the OSD tab in configurator shows.
4.  Compare it to what OSD in the analog goggles show -  Completely different

## Expected behavior
Expected that what we set up in OSD tab will match what we see in the goggles

## Suggested solution(s)
identify and fix the issue

## Additional context
None

---
- FC Board name and vendor: 
- INAV version string:
# INAV/FLYWOOF745NANO 7.1.1 Apr 30 2024 / 20:50:21 (523c4839)
# GCC-10.3.1 20210824 (release)

MrD-RC, 2024-05-02:
You need to update the font. The button is at the bottom of the OSD page.

wernerterreb, 2024-05-02:
Thank you.   I did not know that because I am still new to INAV