iNavFlight/inav #10001: Osd doesn't switch to HD with HDZERO

Pascal687, 2024-05-02:
OSD PROBLEM

Speedybee FC on Inav 7.1 with HDZERO Race1 Vtx and Sharbyte VRX on FS HDO2.

-Bad display both fonts and position (see photo)
-When activated: Inav menu's display is good
-When activated: VTX menu is bad same issue as main display
-I have another plane with the same FC with good osd (see photo). Have swapped the VTX cam with the same results (seems not VTX related)
-Have tried WALKSNAIL on the same FC, no display problem.
-Have mounted another FC MATEK F405 with the same issue.
-Same problem view here https://youtu.be/q_HLM5wrEBQ?si=ZOotPrHq8b35rthH

[image]
 

[image]

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

MrD-RC, 2024-05-03:
Are you sure you're using the correct font in HDZero?

Both Walksnail and HDZero are using the same code to make and provide the OSD. So if Walksnail is working fine. It is unlikely an INAV issue.

Pascal687, 2024-05-03:
Hi Darren.

I’m gonna re check this . I probably have a video too.
Kind regards 
Pascal

Pascal687, 2024-05-03:
Yesterday I managed to get it on 6.0.0 fw while opening Inav menu... but it did not last after the menu closed

https://youtu.be/f5uv4LElwa8

I tried again the FC on 7.1 with walksnail without any problem but any improvement in HDZero (to be sure I added a font file in the Ressource file)

https://github.com/iNavFlight/inav/assets/116525299/1a1f99cf-ca5e-4e13-b1c7-9d0be0ef3ffc

Pascal687, 2024-05-03:
Same HDZero VRX with another bird on SPEEDYBEE with 6.0.0 fw... no problem

I tried to downgrade the fw on the other one but the problem is the same.

https://github.com/iNavFlight/inav/assets/116525299/e4ceaf82-f6fb-42a2-9aac-c2bd17d367d3

Pascal687, 2024-05-06:
Solved; Problem due to overlapping elements !