iNavFlight/inav #10003: Flashed INAV 7.1.0 to speedybee f405 and had issues on quad crash, the motors wouldn't stop.

Al05ex, 2024-05-03:
## Current Behavior

Hy, i have just flashed INAV 7.1.0 to my speedybee 405 stack for my 7 inch long range quad, and i run it for a test flight that resulted in a crash close to my home position (i was inside a building), and i had failsafe set to return to home/them land. The box for alternate option in case of failsafe (close to home by X meters) was unticked, so i set it to DROP and it works now. My problem was that for some reason, my flysky fs i6x with added elrs module, set on 50 hz update rate, was communicatig pretty slow with the quad, the arming/disarmin had a noticeable delay (like seconds from when i disarm and the quad actualy stops motors), so when i crashed, i disarmed but nothing happened and i toggled the switch again, then i suppose it went in a RTH loop cause it didn't responded to my radio commands and motors wouldn't stop until i unplugged the battery. I didn't had a kill switch. On 100hz update rate on elrs work without lagging, the commands are pretty fast. i don't know why on 50 hz i had issues with the signal, or when i am closser to the drone, sometimes the signal intrerupts them comes back on, and so on..

Now, i had another crash in a tree, far from home, my analog video went blur so i didn't see the crash, and initiated failsafe (stage 2 i think is assigned in modes tab) switch on radio for return to home, instead of disarming. I hoped the video would recover and i would start seeing the quad again.. but i also lost radio signal, and the quad motors were spinning at FULL throttle trying to come home i suppose.. It was the second time i had to unplug the battery, and the crash wasn't noticed.. And i didn't disarmed this time, hopping i would see the quad again, but then i couldn't disarm as the radio signal went off and failsafe was initiated (if not already by my switch, then by signal lost).

My question, do i have to make a setting in INAV, or why wouldn't the quad being flipped upside down, at full throttle and vibrating like crazzy while hitting the ground, trigger some sort of auto disarming/killing the drone power? The accelerometer would notice that the quad is flipped, and adding the barometer and the gps, that it doens't respond to the commands and it is not moving in the air as it should (it receives the command X and it doen't do anything cause it is immobilized on the ground and propellers melted.. also one motor got too hot and melted the enamel wire causing a short a guess, cause it doesn't work anymore..)..

I have a video with the flight here [https://drive.google.com/file/d/1UJ1qviP...sp=sharing](https://drive.google.com/file/d/1UJ1qviPPpc1eyb3kkN2dO7Jkqd7Y4yeD/view?usp=sharing)
also log files from blackbox..  [https://drive.google.com/file/d/1n3eGJ5g...sp=sharing](https://drive.google.com/file/d/1n3eGJ5gmZfpxXCfxPyDUvIr9xVAGQLYU/view?usp=sharing)

Asside to this crash issue, there might be an issue with elrs configuration to work with Inav? i have the latest elrs version installed on my happymodel tx24 / betafpv 2.5 ghz rx, with OpenTx on my flysky i6s radio. I have seen there are telemetry update rates for the RX in the lua script (on my radio screen, where i can access the RX settings), that was set at F1000 initially, then i changed it to match the radio, to 100hz.

## Steps to Reproduce

1.Crashed the quad
2.Pressed the failsafe switch, hopping it will come home (i didn't see the crash because video interrupted in that moment)
3.waited, but then lost radio signal, so i couldn't disarm now
4.found the quad spinning at full throttle and had to unplug the battery

## Expected behavior

The quad to disarm because it crashed, and bepper/finder buzzer start beeping waiting for me to find the quad somewhere in the field.

## Suggested solution(s)

A detection in case of a crash, that kills the quad, would be nice. To not spin the motors, hurting someone that have to unplug the battery. There is a kill switch, but if radio is off, and the drone is verry far crashed somewhere, the initialization of RTH function will overheat the motors that try to spin being immobilized, it should disarm when it dosn't act as intended by the flight controller.

## Additional context

https://pastebin.com/NfHn6NsZ

---

- FC Board name and vendor:
# INAV/SPEEDYBEEF405V3 7.1.0 Mar 27 2024 / 14:33:01 (59a6ee61)
# GCC-10.3.1 20210824 (release)
- INAV version string:

[LOG00040.TXT](https://github.com/iNavFlight/inav/files/15205518/LOG00040.TXT)
[LOG00039.TXT](https://github.com/iNavFlight/inav/files/15205519/LOG00039.TXT)
[LOG00038.TXT](https://github.com/iNavFlight/inav/files/15205520/LOG00038.TXT)
[LOG00037.TXT](https://github.com/iNavFlight/inav/files/15205521/LOG00037.TXT)

sensei-hacker, 2024-05-03:
That kinda sounds like you found out why the ELRS documentation keeps telling you to put Arm on channel 5.

Also, don't set ELRS to send many more channels than you need, or at a much higher resolution than you need.  Any channel other than 5 is slower to disarm. Sending more channels is slower. Sending them with higher resolution is slower.

MrD-RC, 2024-05-04:
This is an ELRS/ELRS setup issue. 

For a start. You have arming on channel 8. It needs to be on channel 5 with ELRS, and in a high (1500 and above) for armed. 

INAV does not throttle communications or slow down telemetry. That is all down to the communications system and their protocol.

Al05ex, 2024-05-04:
> That kinda sounds like you found out why the ELRS documentation keeps telling you to put Arm on channel 5.
> 
> Also, don't set ELRS to send many more channels than you need, or at a much higher resolution than you need. Any channel other than 5 is slower to disarm. Sending more channels maker it even slower. Sending them with higher resolution is makes it slower still.

ok, i get it. I have moved my big switch to the right side of my radio so i can disarm with my right hand, and i had to use channel 8 for that. I have configured in mixers tab, that SBA and SBD are reversed, so i moved my switch to channel 5 and the other one (left side switch normally used for arm/disarm) on channel 8. I hope that will do, they are showing as they should in inav.

As for the numbers of channels, i had set in "Switch Mode", under expresslrs tab on my radio, to 16ch rate. Now i selected 12ch, but in inav "receiver" tab still shows up to 17 channels. My radio has 10 available buttons in total that i'd like to use. Or is that actually the resolution..

sensei-hacker, 2024-05-04:
> As for the numbers of channels, i had set in "Switch Mode", under expresslrs tab on my radio, to 16ch rate. Now i selected 12ch, but in inav "receiver" tab still shows up to 17 channels. My radio has 10 available buttons in total that i'd like to use. Or is that actually the resolution..

Yeah that's normal - Configurator / INAV doesn't need to know how many channels will be sent.

It sounds like you figured out how in OpenTx / EdgeTx you can assign any channel to any switch?

Al05ex, 2024-05-04:
> This is an ELRS/ELRS setup issue.
> 
> For a start. You have arming on channel 8. It needs to be on channel 5 with ELRS, and in a high (1500 and above) for armed.
> 
> INAV does not throttle communications or slow down telemetry. That is all down to the communications system and their protocol.
> 
> Also, you have a switch set to failsafe. You should not have this. The failsafe mode is for testing the failsafe procedure only. Not for day-to-day flying.
> 
> Instead, you should use the appropriate flight mode. For example RTH if you want it to return home. Or just disarm if you want it to drop.

i have changed the channels now i mixers tab, to reverse the switches i wanted to use (right switch for arm/disarm).

As for return to home, how can i trigger it manually by pressing a switch? in case i want the quad to come home if i loose video or something..
I had that failsafe switch that did just that, but simulated a real failsafe i guess. In betaflight there is a gps rescue option in modes tab.

Excluding the channels mismatches, still in case of both failsafes, triggered by failsafe switch or by real loose of radio link, my quad initiated a return to home procedure, i guess, and run the motors at full throttle trying to get back home, even tho it was crashed in a tree upside down. That is the real issue, that even after manny seconds/ or minutes if i had let it, it didn't disarmed and didn't knew it was crashed, but still it was trying to get home. This is leading to an overheating of the motors, because they won't stop.

Isn't there a procedure that recognizes the quad not responding/not moving as it should and kill the power?

Al05ex, 2024-05-04:
> Yeah that's normal - Configurator / INAV doesn't need to know how many channels will be sent.
>
> It sounds like you figured out how in OpenTx / EdgeTx you can assign any channel to any switch?

yes, i just saw that i can move the switch to channel 5, without having to actually plug it's wires in another socket on the motherboard..:))

sensei-hacker, 2024-05-04:
> I had that failsafe switch that did just that, but simulated a real failsafe i guess. In betaflight there is a gps rescue option in modes tab.

In INAV, there's a Return to Home mode in the modes tab. It shows up after you have a GPS set up. If you looked at modes before setting up GPS, that would explain why you didn't see it.

Al05ex, 2024-05-04:
> In INAV, there's a Return to Home mode in the modes tab. It shows up after you have a GPS set up. If you looked at modes before setting up GPS, that would explain why you didn't see it.

Amm.. i have 12 sats right now, but still i can't see the RTH, only failsafe in modes tab.. I have air mode off, may that be an issue?

Al05ex, 2024-05-04:
OK, after rebooting in angle mode, i see all those navigation modes option now.

Wich one of these do i need to "park" the quad in the air and hold still ? i had set up NAV ALTHOLD + HEADING HOLD, but because i didn't had a magnetometer, only the althold worked. What do NAV POSHOLD and NAV COURSE HOLD do then ?

breadoven, 2024-05-06:
Have to admit it would be useful to have an auto disarm if the quad has flipped on its back on the ground and is destroying it's motors. Shouldn't be too difficult to define the logic for this without risking false disarms during flips etc. How long can any quad spend upside down in normal flight with no vertical speed ?

Al05ex, 2024-05-06:
https://www.youtube.com/watch?v=NZS0vmy4Cqs

i have uploaded the video with the crash on yt, to better see the issue here.

sensei-hacker, 2024-05-06:
> Wich one of these do i need to "park" the quad in the air and hold still ? i had set up NAV ALTHOLD + HEADING HOLD, but because i didn't had a magnetometer, only the althold worked. What do NAV POSHOLD and NAV COURSE HOLD do then ?

That's Position mode. "Position" in this case means "location".  With the sticks centered, it will maintain the current position. Push the stick forward z it will fly forward. Stick left, it will go left.

Al05ex, 2024-05-06:
ok ,thanks. I will try when i get my quad back i the air :))

MrD-RC, 2024-05-06:
> https://www.youtube.com/watch?v=NZS0vmy4Cqs
> 
> 
> 
> i have uploaded the video with the crash on yt, to better see the issue here. The motors would spin indefinitely, until i unplug the battery, this lead to a motor overheating and won't work anymore.
> 
> 
> 
> By the way, thanks for your help so far, with the arming buttons. I am still a begginer in 7 inch world, and fpv in general...

That video doesn't help at all. It just shows that you crashed and the motors are still running. There is no OSD to show anything that may be slightly useful.

Al05ex, 2024-05-06:
> That video doesn't help at all. It just shows that you crashed and the motors are still running. There is no OSD to show anything that may be slightly useful.

unfortunately it seems like i haven't recorded the OSD.. but i have posted the log files from sd card, if those help with anything. i don't know how to interpret them in blackbox viewer.