iNavFlight/inav #10009: INAV 7.1.0 failed to turn off motors when disarm the quadcopter

capblack, 2024-05-05:
Hello, I have a quad with Matek 722-SE that has already flown about five times very well and stable with a lot of stability, the position and altitude hold, today when I wanted to fly it it shot out of control and I immediately used the switch to disarm it, it started to go down but the engines did not turn off as expected and it fell backwards with the engines at maximum, bouncing and jumping across the floor, even causing me some cuts on my thigh, until it stopped when the battery came off. I want to comment on this because it bothered me. It seemed very strange that the engines were not turned off, the only thing I can add is a video with what happened because the FC did not have a memory card at that time

https://youtu.be/KZ1lo_wYR_I

capblack, 2024-05-06:
I would like to add that I am using a DJI remote with original air unit.

sensei-hacker, 2024-05-06:
In the video it looks to me like the props stopped?
Disarming does take 0.1 seconds.

sensei-hacker, 2024-05-06:
Also, could you please attach your diff.

capblack, 2024-05-06:
Hello sensei-hacker, believe me when I tell you that it fell with the motors running and began to jump and bounce until the battery was fired and it separated from the drone. If you pay attention, the left arm seems stopped but the right one can be seen rotating. I'm attaching it too. a photo of my thigh a stopped engine can't do this

[INAV_7.1.0_cli_20240506_202528.txt](https://github.com/iNavFlight/inav/files/15224439/INAV_7.1.0_cli_20240506_202528.txt)

[image]

sensei-hacker, 2024-05-06:
I believe you. Just double checking because the video *looked* stopped - but the limited frame rate of video can do that.

Got a diff?

capblack, 2024-05-06:
[INAV_7.1.0_cli_20240506_202528.txt](https://github.com/iNavFlight/inav/files/15225530/INAV_7.1.0_cli_20240506_202528.txt)

this file is the diff

breadoven, 2024-05-07:
Not being able to disarm is usually caused by being in failsafe. Was there any indication failsafe was active which would have triggered a RTH (probably RTH landing given it was at the home position) ?

capblack, 2024-05-07:
I have watched the video again in slow motion and it seems that if the motors stop until it hits the floor and there they are activated again and it flip over. After that I heard the motors working and I could feel it hit in many places until I felt it hit the thigh and then stopped when the battery came off

breadoven, 2024-05-07:
You're right, the motors did stop spinning in the video then started again when it landed upright on the floor. So it did disarm but then somehow managed to rearm. Are you sure you didn't accidently rearm again ?

capblack, 2024-05-07:
I doubt I was able to activate it again, the only thing I can add is that in addition to the propellers damaged, two GPS/MAG cables were cut, well I guess I'll repair it and try again in an open space to see how it behaves.
The only other change was to use a 1550 mAh battery instead of the 1800 with 20% less weight
I don't know if that could explain the behavior.

capblack, 2024-05-08:
I don't know, I can't guarantee this. I checked the DJI log and until the last moment the battery was disconnected, they reported that the RC was on ch 4.
"93
00:00:13,665 --> 00:00:13,813
signal:3 ch:65535 flightTime:6 uavBat:14.0V glsBat:10.9V uavBatCells:4 glsBatCells:3 delay:34ms bitrate:21.2Mbps rcSignal:4"
Unfortunately it's the only thing I have for reference.

mmosca, 2024-05-09:
> I don't know, I can't guarantee this. I checked the DJI log and until the last moment the battery was disconnected, they reported that the RC was on ch 4. "93 00:00:13,665 --> 00:00:13,813 signal:3 ch:65535 flightTime:6 uavBat:14.0V glsBat:10.9V uavBatCells:4 glsBatCells:3 delay:34ms bitrate:21.2Mbps rcSignal:4" Unfortunately it's the only thing I have for reference.
> 
> but in fact the motors are activated when it hits the ground, I was also able to observe with filmora9 that when it is falling I have my hands on the sticks

are you using dji rc?

capblack, 2024-05-09:
yes I am using a DJI FPV Remote Controller V1

capblack, 2024-05-15:
now that I think about it . That the drone has started to rise uncontrollably and that it could be due to interference from satellites here in Israel? Since the only thing I had activated was PH and that INAV 7.1, which does not require the barometer? since before it was always very stable