iNavFlight/inav #10015: Issue with Barometer DPS310 Activation on SpeedyBee F7 V3 Stack

borattoo, 2024-05-08:
## Current Behavior

When activating the GEPRC M1025-DQ GPS, DPS310 barometer on my SpeedyBee F7 V3 stack with iNav 7.1.1, a red icon appears next to the barometer status and the ESC continuously beeps the motors, preventing me from completing the configuration of my quad.
## Steps to Reproduce

1. Flash iNav 7.1.1 on SpeedyBee F7 V3 stack.
2. Install Geprc M1025-DQ GPS,
3. Activate GPS and magnetometer successfully.
4. Enable DPS310 barometer.
5. Observe the error behavior with red icon and beeping motors.

## Expected behavior

I expected the barometer to activate without causing any errors or abnormal ESC behavior, allowing for complete configuration of the quad.
## Suggested solution(s)

I am unsure of the specific cause of the issue, so any guidance on troubleshooting or settings adjustments that might resolve the conflict between the barometer and other components would be helpful.
## Additional context

[image]
[image]

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/SPEEDYBEEF7V3 7.1.1 May  6 2024 / 12:53:44 (dd91a871)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# resources

# Timer overrides
timer_output_mode 0 AUTO
timer_output_mode 1 MOTORS
timer_output_mode 2 MOTORS
timer_output_mode 3 MOTORS
timer_output_mode 4 AUTO
timer_output_mode 5 AUTO
timer_output_mode 6 AUTO
timer_output_mode 7 AUTO
timer_output_mode 8 AUTO
timer_output_mode 9 AUTO
timer_output_mode 10 AUTO
timer_output_mode 11 AUTO
timer_output_mode 12 AUTO
timer_output_mode 13 AUTO

# Outputs [servo]
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
[... 1828 lines cut ...]
set nav_fw_min_thr = 1200
set nav_fw_max_thr = 1700
set nav_fw_pitch2thr = 10
set nav_fw_launch_thr = 1700
set nav_fw_launch_idle_thr = 1000
set limit_cont_current = 0
set limit_burst_current = 0
set limit_burst_current_time = 0
set limit_burst_current_falldown_time = 0
set limit_cont_power = 0
set limit_burst_power = 0
set limit_burst_power_time = 0
set limit_burst_power_falldown_time = 0

# save configuration
save

# end the command batch
batch end

# 
---

- FC Board name and vendor: SpeedyBee F7 v3
- INAV version string: 7.1.1

MrD-RC, 2024-05-08:
The issue is that GNSS module has a barometer on it. It is conflicting with the barometer on the flight controller. Ideally, you should just use a GNSS module without a barometer. Such as a Matek M10 or the Walksnail M181 M10. The other option is to physically remove the barometer from the GNSS module. 

Avoid the GepRC modules unless your flight controller doesn't have a barometer.

closed 2024-05-08 by MrD-RC

borattoo, 2024-05-08:
Thank you for the quick response, @MrD-RC.

I understand the issue with the GNSS module's barometer conflicting with the one on the flight controller. Currently, I do not have access to a GNSS module without a barometer. Is there a way to disable the barometer on the GNSS module through CLI commands or configuration settings in iNav? This would allow me to continue using my current setup without needing to physically modify the hardware.

I appreciate any further guidance you can provide.

MrD-RC, 2024-05-08:
No. INAV doesn't have control over the baro on the GNSS. The two baros are conflicting. So no way to talk to them individually.

borattoo, 2024-05-08:
so better use a gps without barometer and compass? i will get all functionality dont using 

> No. INAV doesn't have control over the baro on the GNSS. The two baros are conflicting. So no way to talk to them individually.

Thank you for the clarification, @MrD-RC.

Given that INAV cannot control the barometer on the GNSS module individually, would it be advisable to use a GPS unit that only connects via RX, TX, Ground, and Power, effectively foregoing the magnetometer as well? Would I still be able to fully utilize INAV's features without the magnetometer?

Additionally, I'm aware that Betaflight has a set baro_i2c_device command that allows users to select which barometer device to use. Is there a similar command or any workaround in INAV that could help resolve this issue without having to replace the GPS module?

I appreciate your guidance on how best to configure my setup under these circumstances.

sensei-hacker, 2024-05-08:
> Would I still be able to fully utilize INAV's features without the magnetometer?

You can disconnect the I2C and forego the mag. INAV works fairly well without a mag. When operating without a compass, it's a good idea to fly straight forward for a few meters on take off so that INAV can figure out which direction is forward.

> baro_i2c_device command

Unfortunately, when you have two of the same barometer attached to the same pin, there is no way to distinguish them. You have both on bus 2.  The baro_i2c_device command specifies which bus the baro is on.  With twin baros on the same bus, they are indistinguishable.  Telling it not to use that bus would mean neither copy of the baro is reachable.

Some people have used a very fine tip on their soldering iron, and a magnifying glass, to disable the baro on such a combined GPS / baro board. That can of course go badly.