iNavFlight/inav #10017: iNav custom target for servo on LED pin

masterguy29, 2024-05-10:
***Apologies to the PC folk if the formatting is bad, I am on mobile.***

I have a Tmotor Velox Cine F7(TMOTORVELOXF7V2) that I have built. I have a gimbal on my drone that allows me to move my camera up or down via a switch on my transmitter. 

I had betaflight on it and was easily able to remap my LED pins to be used as a servo output. However it's not so easy on iNav. I have pulled the C and H files for my target board but I don't know what to change in order to make this work correctly without messing anything up. Any help would be greatly appreciated.

I know how to compile the files into hex for firmware flashing, I just need to know what to change to make my LED pin usable for my gimbal.

Here is my C file:

timerHardware_t timerHardware[] = {

 DEF_TIM(TIM3, CH1, PB4,   TIM_USE_OUTPUT_AUTO, 0, 0),   // S1   UP1-2   D(1, 4, 5)
 
 DEF_TIM(TIM3, CH2, PB5,   TIM_USE_OUTPUT_AUTO, 0, 0),   // S2   UP1-2   D(1, 5, 5)

 DEF_TIM(TIM3, CH3, PB0,   TIM_USE_OUTPUT_AUTO, 0, 0),   // S3   UP1-2   D(1, 7, 5)

 DEF_TIM(TIM3, CH4, PB1,   TIM_USE_OUTPUT_AUTO, 0, 0),   // S4   UP1-2   D(1, 2, 5)

 DEF_TIM(TIM2, CH1, PA15,  TIM_USE_OUTPUT_AUTO, 0, 0),   // S5   UP1-7   D(1, 5, 3) - clash with S2

 DEF_TIM(TIM2, CH2, PB3,   TIM_USE_OUTPUT_AUTO, 0, 0),   // S6   UP1-7   D(1, 6, 3)
 
 DEF_TIM(TIM4, CH1, PB6,   TIM_USE_OUTPUT_AUTO, 0, 0),   // S7   UP1-6   D(1, 0, 2) // used to be fw motor
 
 DEF_TIM(TIM4, CH2, PB7,   TIM_USE_OUTPUT_AUTO, 0, 0),   // S8   UP1-6   D(1, 3, 2) // used to be fw motor

 DEF_TIM(TIM1, CH1, PA8,     TIM_USE_LED,    0,  1), // LED
};
_____________________________________

I know I need to delete "DEF_TIM(TIM1, CH1, PA8,     TIM_USE_LED,    0,  1), // LED" and replace one of the timers to use PA8 instead, I'm just not sure which one to change. 
_____________________________________

Here is my H file:

\#pragma once

\#define TARGET_BOARD_IDENTIFIER "TVF7"

\#define USBD_PRODUCT_STRING  "TMOTORVELOXF7V2"

\#define LED0                    PA14

\#define LED1                    PA13

\#define BEEPER                  PC13

\#define BEEPER_INVERTED

\// SPI

\#define USE_SPI

\#define USE_SPI_DEVICE_1

\#define SPI1_SCK_PIN            PA5

\#define SPI1_MISO_PIN           PA6

\#define SPI1_MOSI_PIN           PA7

\#define USE_SPI_DEVICE_2

\#define SPI2_SCK_PIN            PB13

\#define SPI2_MISO_PIN           PB14

\#define SPI2_MOSI_PIN           PB15

\#define USE_SPI_DEVICE_3

\#define SPI3_SCK_PIN            PC10

\#define SPI3_MISO_PIN           PC11

\#define SPI3_MOSI_PIN           PC12

\//GYRO

\#define USE_IMU_ICM42605

\#define IMU_ICM42605_ALIGN      CW270_DEG

\#define ICM42605_CS_PIN         PB2

\#define ICM42605_SPI_BUS        BUS_SPI1

\// *************** I2C /Baro/Mag *********************

\#define USE_I2C

\#define USE_I2C_DEVICE_1

\#define I2C1_SCL                PB8

\#define I2C1_SDA                PB9

\#define USE_BARO

\#define BARO_I2C_BUS            BUS_I2C1

\#define USE_BARO_ALL

\#define USE_MAG

\#define MAG_I2C_BUS             BUS_I2C1

\#define USE_MAG_ALL

\#define TEMPERATURE_I2C_BUS     BUS_I2C1

\#define PITOT_I2C_BUS           BUS_I2C1

\#define RANGEFINDER_I2C_BUS     BUS_I2C1

\// *************** SPI2 OSD ***********************

\#define USE_MAX7456

\#define MAX7456_SPI_BUS         BUS_SPI2

\#define MAX7456_CS_PIN          PB12

\// *************** SPI3 SD BLACKBOX*******************

\#define USE_FLASHFS

\#define USE_FLASH_M25P16

\#define M25P16_SPI_BUS          BUS_SPI3

\#define M25P16_CS_PIN           PD2

\#define ENABLE_BLACKBOX_LOGGING_ON_SPIFLASH_BY_DEFAULT

\// *************** UART *****************************

\#define USE_VCP

\#define USE_UART1

\#define UART1_TX_PIN            PA9

\#define UART1_RX_PIN            PA10

\#define USE_UART2

\#define UART2_TX_PIN            PA2

\#define UART2_RX_PIN            PA3

\#define USE_UART3

\#define UART3_TX_PIN            PB10

\#define UART3_RX_PIN            PB11

\#define USE_UART4

\#define UART4_TX_PIN            PA0

\#define UART4_RX_PIN            PA1

\#define USE_UART6

\#define UART6_TX_PIN            PC6

\#define UART6_RX_PIN            PC7

\#define SERIAL_PORT_COUNT       6

\#define DEFAULT_RX_TYPE         RX_TYPE_SERIAL

\#define SERIALRX_PROVIDER       SERIALRX_SBUS

\#define SERIALRX_UART           SERIAL_PORT_USART1

\// *************** ADC *****************************

\#define USE_ADC

\#define ADC_INSTANCE                ADC1

\#define ADC1_DMA_STREAM             DMA2_Stream0

\#define ADC_CHANNEL_1_PIN           PC2

\#define ADC_CHANNEL_2_PIN           PC1

\#define ADC_CHANNEL_3_PIN           PC0

\#define VBAT_ADC_CHANNEL            ADC_CHN_1

\#define CURRENT_METER_ADC_CHANNEL   ADC_CHN_2

\#define RSSI_ADC_CHANNEL            ADC_CHN_3

\// *************** LEDSTRIP ************************

\#define USE_LED_STRIP

\#define WS2811_PIN                  PA8

\#define DEFAULT_FEATURES                (FEATURE_OSD | FEATURE_CURRENT_METER | FEATURE_VBAT | FEATURE_TX_PROF_SEL | FEATURE_BLACKBOX)

\#define CURRENT_METER_SCALE     250

\#define USE_SERIAL_4WAY_BLHELI_INTERFACE

\#define TARGET_IO_PORTA 0xffff

\#define TARGET_IO_PORTB 0xffff

\#define TARGET_IO_PORTC 0xffff

\#define TARGET_IO_PORTD 0xffff

\#define MAX_PWM_OUTPUT_PORTS        8

\#define USE_DSHOT

\#define USE_SERIALSHOT

\#define USE_DSHOT_DMAR

\#define USE_ESC_SENSOR
_____________________________________

On this one I know I need to delete "// \*************** LEDSTRIP ************************\
\#define USE_LED_STRIP
\#define WS2811_PIN                  PA8" entirely. 

Any help would be greatly appreciated! 

Thank you!

sensei-hacker, 2024-05-10:
I see the standard target already has eight outputs.  Are you using all eight for eight motors?

masterguy29, 2024-05-10:
> I see the standard target already has eight outputs. Are you using all eight for eight motors?

No, I am only using 4 motors. Standard X Quad.

sensei-hacker, 2024-05-10:
In target.c, you could replace TIM_USE_LED with TIM_USE_OUTPUT_AUTO

Whether that's simpler than using an existing output ... :)

masterguy29, 2024-05-10:
I will try that and report back. Thanks!

And yeah I know I'm just too lazy to resolder the gimbal to something else lol....

masterguy29, 2024-05-11:
I have figured it out and gotten it working. I will attach text files of my Target C and Target H for the TMOTORVELOXF7V2 target. This config will work if you would like to use a LED pin for a servo instead of an LED strip. Config has been tested on the bench and servo and motors all work correctly.
[Target_C.txt](https://github.com/iNavFlight/inav/files/15281105/Target_C.txt)
[Target_H.txt](https://github.com/iNavFlight/inav/files/15281106/Target_H.txt)

closed 2024-05-11 by masterguy29