iNavFlight/inav #10018: FLYWOO F745 motor mapping incorrect in iNav version 7

wsiok, 2024-05-10:
## Current Behavior
One of Motor mappings with the target FLYWOOF745 uses port E13.

## Steps to Reproduce
1. Flash the iNav version 7.

## Expected behavior
The motors shall be mapped as follows according to the documentation. iNav 5 does it correctly.
https://flywoo.tawk.help/article/goku-f745-2-6s-16x16-stack 
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 E09
resource MOTOR 4 E11

## Suggested solution(s)
See the diff beneath

---
iNav version
7.1.1
- FC Board name and vendor: FLYWOO F745 

Workaround / fix:

diff --git a/src/main/target/FLYWOOF745/target.c b/src/main/target/FLYWOOF745/target.c
index 04c083ac9..85b572d54 100644
--- a/src/main/target/FLYWOOF745/target.c
+++ b/src/main/target/FLYWOOF745/target.c
@@ -30,26 +30,17 @@
 #include "drivers/pinio.h"
 
 timerHardware_t timerHardware[] = {
-// DEF_TIM(TIM1, CH3, PE13, TIM_USE_PPM,                      0, 1), // PPM,
-// DMA2_ST6
+    //DEF_TIM(TIM1, CH3, PE13, TIM_USE_PPM,                      0, 1), // PPM, DMA2_ST6
 
-#ifdef FLYWOOF745
-    DEF_TIM(TIM3, CH4, PB1,  TIM_USE_OUTPUT_AUTO, 0, 0),   // M1 , DMA1_ST2
-    DEF_TIM(TIM8, CH4, PC9,  TIM_USE_OUTPUT_AUTO, 0, 0),   // M2 , DMA2_ST7
-    DEF_TIM(TIM3, CH3, PB0,  TIM_USE_OUTPUT_AUTO, 0, 0),   // M3 , DMA1_ST7
-    DEF_TIM(TIM1, CH2, PE11, TIM_USE_OUTPUT_AUTO, 0, 1),  // M4 , DMA2_ST4
-    DEF_TIM(TIM1, CH1, PE9,  TIM_USE_OUTPUT_AUTO, 0, 2),   // M5 , DMA2_ST2
-#else /* FLYWOOF745NANO */
-    DEF_TIM(TIM3, CH3, PB0,  TIM_USE_OUTPUT_AUTO, 0, 0),   // M1 , DMA1_ST7
-    DEF_TIM(TIM3, CH4, PB1,  TIM_USE_OUTPUT_AUTO, 0, 0),   // M2 , DMA1_ST2
-    DEF_TIM(TIM1, CH1, PE9,  TIM_USE_OUTPUT_AUTO, 0, 2),   // M3 , DMA2_ST2
-    DEF_TIM(TIM1, CH2, PE11, TIM_USE_OUTPUT_AUTO, 0, 1),  // M4 , DMA2_ST4
-    DEF_TIM(TIM8, CH4, PC9,  TIM_USE_OUTPUT_AUTO, 0, 0),   // M5 , DMA2_ST7
-#endif
-    DEF_TIM(TIM5, CH4, PA3,  TIM_USE_OUTPUT_AUTO, 0, 0),  // M6 , DMA1_ST1
-    DEF_TIM(TIM3, CH1, PB4,  TIM_USE_OUTPUT_AUTO, 0, 0),  // M7 , DMA1_ST4
-    DEF_TIM(TIM3, CH2, PB5,  TIM_USE_OUTPUT_AUTO, 0, 0),  // M8 , DMA1_ST5
-    DEF_TIM(TIM4, CH1, PD12, TIM_USE_LED, 0, 0),         // LED_STRIP, DMA1_ST0
+    DEF_TIM(TIM3, CH3, PB0,  TIM_USE_MOTOR,                    0, 0), // M1 , DMA1_ST7
+    DEF_TIM(TIM3, CH4, PB1,  TIM_USE_MOTOR,                    0, 0), // M2 , DMA1_ST2
+    DEF_TIM(TIM1, CH1, PE9,  TIM_USE_MOTOR,    0, 2), // M3 , DMA2_ST2
+    DEF_TIM(TIM1, CH2, PE11, TIM_USE_MOTOR,    0, 1), // M4 , DMA2_ST4
+    //DEF_TIM(TIM8, CH4, PC9,  TIM_USE_SERVO,    0, 0), // M5 , DMA2_ST7
+    //DEF_TIM(TIM5, CH4, PA3,  TIM_USE_SERVO,    0, 0), // M6 , DMA1_ST1
+    //DEF_TIM(TIM3, CH1, PB4,  TIM_USE_SERVO,    0, 0), // M7 , DMA1_ST4
+    //DEF_TIM(TIM3, CH2, PB5,  TIM_USE_SERVO,    0, 0), // M8 , DMA1_ST5
+    //DEF_TIM(TIM4, CH1, PD12, TIM_USE_LED,                                               0, 0), // LED_STRIP, DMA1_ST0
 };
 
 const int timerHardwareCount = sizeof(timerHardware) / sizeof(timerHardware[0]);

damianmroz, 2024-07-04:
Have the same issue. Flashed inav 7.1.2 today bug still exists. Reverted back to 6.0 don’t know how to solve that.

sensei-hacker, 2024-07-04:
To get the old mapping, set timer 1 and timer 3 to be motors, but not timer 8.

The new mapping has S2 on what used to be S5.

Nandox7, 2024-07-06:
Having the same issue.
Can the remapping be done using the CLI or I need to compile a new version of the FW?

Minutedrone, 2024-07-16:
Hello, 

I've had this issue and reported this to flywoo.
They gave me a specific inav firmware for this target which can be found here :
https://drive.google.com/drive/folders/1IKJbPK2We3eIKuv441stdowBpwglx665

Everything works fine after flashing this firmware.
I'm uploading it here so someone can update the firmware and correct the target.

[inav_7.1.0_FLYWOOF745.hex.zip](https://github.com/user-attachments/files/16252884/inav_7.1.0_FLYWOOF745.hex.zip)