iNavFlight/inav #10024: Why doesn't vtx catch a larger range of video signal? (pit)

IonescuSergiuMihai, 2024-05-12:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

IonescuSergiuMihai, 2024-05-12:
Hello, I have a vtx that does not want to catch a larger range, maximum 2 meters. And I know it would be on pit mode.

vtx: tx805

I use the INAV system

sensei-hacker, 2024-05-12:
A VTX that only goes 2 meters means the channels don't match, or one of the antennas is broken / disconnected, unless you're in out mode.

Check the instructions on your VTX and look at the LEDs to see a) if you're in pit mode and b) what channel you are on.

anti-vaxxer, 2024-05-12:
unfortunately pit mode doesn't work in INAV(see issue #8369) so you can't disable pit mode with INAV.
to disable pit mode you have to use your VTX's physical button if it has one to get out of pit mode

IonescuSergiuMihai, 2024-05-12:
[image]

[image]

 I don't know when pit mode is on except that it gives me a very weak signal, and I tried what the instructions said.

Plus the antennas are ok and new.

And I changed channels and bands from time to time.

anti-vaxxer, 2024-05-12:
do any of the 3 red, green or blue LEDs light up or flicker? I understood from the instructions that all of them are off when in pit mode.

instructions say to press button for 12 seconds to exit pit mode. if that doesn't get it out of pit mode there might be something wrong with the VTX or something else.

I'm not sure whether it helps but you can try disconnecting the SmartAudio wire and then pressing the button again.

IonescuSergiuMihai, 2024-05-12:
it did not work