iNavFlight/inav #10026: Servos not smooth

ammikey, 2024-05-13:
****
Matek 405 TE SD Vtol Flight controller. ELRS System 

## Current Behavior
Everything is working correctly except servo outputs 10 and 11. The servos are moving in large steps and not smooth.
Signal from the transmitter is smooth and linier but the reception in the INAV Configurator receiver window shows a very choppy signal in. All other servos and outputs are working correctly. 

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior
Servos to work smoothly

## Suggested solution(s)
 Unknown

## Additional context

I couldn't Get this to work. 

---

 Version
# INAV/MATEKF405TE_SD 7.1.0 Mar 27 2024 / 13:54:51 (59a6ee61)
# GCC-10.3.1 20210824 (release)

- FC Board name and vendor: MATEKF405TE_SD. GetFpv
- INAV version string: INAV/MATEKF405TE_SD 7.1.0 Mar 27 2024 / 13:54:51 (59a6ee61)
# GCC-10.3.1 20210824 (release)

sensei-hacker, 2024-05-13:
Most likely you're using ELRS in a switch mode rather than full resolution mode. Higher channels will only have eight or sixty four steps if you use a switch mode in ELRS.

https://www.expresslrs.org/software/switch-config/#summary-of-switch-configuration-modes

ammikey, 2024-05-13:
Thank you for your quick response, I'll give that a try tomorrow.
Much Thanks Mike J.

closed 2024-05-13 by MrD-RC