iNavFlight/inav #10043: Wrong Altitude Calculation Aocoda-RC F405 40A AIO DPS310 / SPL06

Developer-MR01, 2024-05-15:
## Current Behavior
- There is no constant altitude level > altitude-value fades up/down and finaly back to the value before.
- The baro counting-step is about "8" (I´m not sure if it´s an issue)

[image]

## Steps to Reproduce
1. default setup FC: Multirotor 3inch 
2. select "AUTO" @ Barometer > Result: SPL06
3. calibrate ACCs
4. final Reboot before starting the test
5. place the Drone on the floor
6. powerOn and wait for "ready to arm"
7. connect to INAV and start "Sensors" > Barometer
8. put the drone up to the desktop or a board (about 1m up)
9. test with and without arm > negativ results
10. test with Releases 6.xx up to 7.11 > negativ results

## Expected behavior
constant altitude level

## Suggested solution(s)

## Additional context

---
INAV/AOCODARCF405AIO 7.1.1 May  6 2024 / 10:54:34 (dd91a871)
GCC-10.3.1 20210824 (release)

MrD-RC, 2024-05-16:
Is the baro covered with open cell foam?

breadoven, 2024-05-16:
When disarmed the Altitude should remain around 0 regardless of height changes of the drone itself. However, when Armed altitude should follow and hold the height changes of the drone ... which it does when I tested just now.

Developer-MR01, 2024-05-16:
> When disarmed the Altitude should remain around 0 regardless of height changes of the drone itself. However, when Armed altitude should follow and hold the height changes of the drone ... which it does when I tested just now.

the test cases again:

ACC calibration completed
Inductive interference (cable) excluded 
(less noise @ACCs)

Test Case #1 (without GPS in Config, Port disabeld)
Drone placed on the ground
Drone connected to battery
Drone NOT armed 
Drone lifted to 2m height (by hand)
> no Altitude value

[image]

Test Case #2 (without GPS Config, Port disabled)
Drone placed on the ground
Drone connected to battery
Drone armed 
Drone lifted to 2m height (by hand)
> Pressure changes after arming (first peak)
> no Altitude value

[image]

Test Case #3 (with GPS)
Drone placed on the ground
GPS configured and connected
Drone shot at battery
Drone NOT armed 
Drone lifted to 2m height (by hand)
> Altitude goes back to original value or is fading

[image]

Test Case #4 (with GPS)
Drone placed on the ground
GPS configured and connected
Drone connected to battery
Drone armed 
Drone lifted to 2m altitude (by hand)
> Pressure changes after arming (first negativ pressure peak)
> Altitude is fading
Note: Scaling has changed due to the large height value

[image]

It seems that only the GPS value is used for the altitude calculation. If the GPS is switched off, the altitude value remains at 0

breadoven, 2024-05-16:
I assume you're using version 7.1.1. If so there's a bug that disables the Baro if `inav_use_gps_no_baro `is set to ON (default setting). Setting to OFF will enable the Baro.

Developer-MR01, 2024-05-16:
same like before...

[image]

[image]

breadoven, 2024-05-16:
> same like before... [...]

Is this armed or disarmed though ?

Developer-MR01, 2024-05-16:
The first test was disarmed.

with armed same result:

[image]

Developer-MR01, 2024-05-16:
I tested it again in the open field today...  fyi: during the tests at a height of 2m, the drone was armed the whole time. The drop in altitude did occur in armed status.